gnuradio/gnuradio #2931: CMake finds codec2 , but swig fails to find it during build

devnulling, 2019-11-28:
Setup:
- Ubuntu 18.04.x 
- GR master
- source installed codec2 (to /usr/local) (current master version of codec2)

During cmake it finds libcodec2:

```
-- Configuring gr-vocoder support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-vocoder support.
--   Override with -DENABLE_GR_VOCODER=ON/OFF
-- Found LIBCODEC2: /usr/local/lib/libcodec2.so  
-- libcodec2 modes: CODEC2_MODE_3200;CODEC2_MODE_2400;CODEC2_MODE_1600;CODEC2_MODE_1400;CODEC2_MODE_1300;CODEC2_MODE_1200;CODEC2_MODE_700;CODEC2_MODE_700B;CODEC2_MODE_700C;CODEC2_MODE_450;CODEC2_MODE_450PWB;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1600_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1600_EN;CODEC2_MODE_3200_EN;CODEC2_MODE_3200_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_2400_EN;CODEC2_MODE_2400_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1600_EN;CODEC2_MODE_1600_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1400_EN;CODEC2_MODE_1400_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1300_EN;CODEC2_MODE_1300_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_1200_EN;CODEC2_MODE_1200_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_700_EN;CODEC2_MODE_700_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_700B_EN;CODEC2_MODE_700B_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_700C_EN;CODEC2_MODE_700C_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_450_EN;CODEC2_MODE_450_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_450PWB_EN;CODEC2_MODE_450PWB_EN;CODEC2_MODE_EN_DEFAULT;CODEC2_MODE_ACTIVE
-- libcodec2 has 700bps mode
-- libcodec2 had FreeDV api modes: FREEDV_MODE_1600;FREEDV_MODE_700;FREEDV_MODE_700B;FREEDV_MODE_2400A;FREEDV_MODE_2400B;FREEDV_MODE_800XA;FREEDV_MODE_700C;FREEDV_MODE_700D;FREEDV_MODE_2020;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_1600_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_1600_EN;FREEDV_MODE_1600_EN;FREEDV_MODE_1600_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_700_EN;FREEDV_MODE_700_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_700B_EN;FREEDV_MODE_700B_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_700C_EN;FREEDV_MODE_700C_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_700D_EN;FREEDV_MODE_700D_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_2400A_EN;FREEDV_MODE_2400A_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_2400B_EN;FREEDV_MODE_2400B_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_800XA_EN;FREEDV_MODE_800XA_EN;FREEDV_MODE_EN_DEFAULT;FREEDV_MODE_2020_EN;FREEDV_MODE_2020_EN;FREEDV_MODE_EN_DEFAULT
-- Found LIBGSM: /usr/lib/x86_64-linux-gnu/libgsm.so  

```

When building swig fails to find it:

```
Scanning dependencies of target vocoder_swig_swig_compilation
[ 57%] Swig source vocoder_swig.i
/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/swig/vocoder_swig.i:79: Error: Unable to find 'codec2/codec2.h'
/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/swig/vocoder_swig.i:102: Error: Unable to find 'codec2/freedv_api.h'
gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/build.make:62: recipe for target 'gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp' failed
make[2]: *** [gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp'
CMakeFiles/Makefile2:13755: recipe for target 'gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/all' failed
make[1]: *** [gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/all] Error 2
```

$ find /usr/local -name "codec2.h"
/usr/local/include/codec2/codec2.h

$ find /usr/local -name "freedv_api.h"
/usr/local/include/codec2/freedv_api.h

michaelld, 2019-11-28:
Please do `make VERBOSE=ON` and then post the actual SWIG command that's causing the error. That should provide critical information on fixing this issue. Guessing it's due to come recent changes to how codec2 is found and used.

devnulling, 2019-11-28:
```
make -f gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/build.make gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/depend
make[2]: Entering directory '/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build'
cd /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/swig /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/DependInfo.cmake --color=
Dependee "/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/DependInfo.cmake" is newer than depender "/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/depend.internal".
Dependee "/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/depend.internal".
Scanning dependencies of target vocoder_swig_swig_compilation
make[2]: Leaving directory '/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build'
make -f gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/build.make gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/build
make[2]: Entering directory '/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build'
[ 56%] Swig source vocoder_swig.i
cd /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig && /usr/bin/cmake -E make_directory /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig.dir
cd /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig && /usr/bin/cmake -E touch /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp
cd /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig && /usr/bin/cmake -E env SWIG_LIB=/usr/share/swig3.0 /usr/bin/swig3.0 -python -fvirtual -modern -keyword -w511 -w314 -relativeimport -py3 -module vocoder_swig -DLIBCODEC2_FOUND -DLIBCODEC2_HAS_FREEDV_API -DLIBGSM_FOUND -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/usr/include/python3.6m -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/lib/pmt/../../include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/volk/include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/volk/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/local/include/codec2 -I/usr/include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/swig -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/swig -I/usr/include/python3.6m -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gr-vocoder/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gnuradio-runtime/lib/../include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/gnuradio-runtime/lib/pmt/../../include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/volk/include -I/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/volk/include -I/usr/include -I/usr/include -I/usr/include -I/usr/include -I/usr/local/include/codec2 -I/usr/include -outdir /home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build/gr-vocoder/swig -c++ -o /home/user/sdr/installs/v3.14.1.1_master/sr [... cut ...]
make[2]: *** Deleting file 'gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp'
make[2]: Leaving directory '/home/user/sdr/installs/v3.14.1.1_master/src/gnuradio/build'
CMakeFiles/Makefile2:13755: recipe for target 'gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/all' failed
make[1]: *** [gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

[make_verbose.txt](https://github.com/gnuradio/gnuradio/files/3902822/make_verbose.txt)

devnulling, 2019-11-28:
@michaelld I'm not sure if this is really worth digging into, I rebuilt the codec2 lib and installed it to `/usr` instead of `/usr/local` and it looks like swig found it.

`cmake -DCMAKE_INSTALL_PREFIX=/usr ..`

michaelld, 2019-11-29:
So the issue is that SWIG is not provided the include directory `/usr/local/include` alone, but instead `/usr/local/include/codec2`, which doesn't help since the `#include`'s are `<codec2/XXX>`.

Also: (2) The `CMAKE_INSTALL_PREFIX` is not set to that for the codec2 headers (`/usr/local`) and hence one of the default directories (`$CMAKE_INSTALL_PREFIX/include`) is not the correct location for those headers.

As you note: Switching both the codec2 and `CMAKE_INSTALL_PREFIX` to `/usr` does the trick because then (2) becomes valid.

So, yes, this is a SWIG bug with respect to the location of `codec2` headers.

mormj, 2022-09-09:
Closing because SWIG is no more beyond 3.8

closed 2022-09-09 by mormj