gnuradio/gnuradio #2938: Resizing crashes: Assertion failed: CAIRO_REFERENCE_COUNT_HAS_REFERENCE (&surface->ref_count), file cairo-surface.c, line 955

segdy, 2019-11-30:
Gnuradio 3.8.0.0 installed on latest Windows 10.
When the window is resized, gnuradio-companion crashes:

```
C:\Program Files\GNURadio-3.8\bin>"C:\Program Files\GNURadio-3.8\bin\run_gr.bat" gnuradio-companion.py
setting gnuradio environment
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.0.0 >>>

Block paths:
        C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks

Loading: "O:\local\test_qt.grc"
>>> Done
Assertion failed: CAIRO_REFERENCE_COUNT_HAS_REFERENCE (&surface->ref_count), file cairo-surface.c, line 955

C:\Program Files\GNURadio-3.8\bin>

```

MavEtJu, 2020-01-25:
Me too!

I have a WAV source and an audio sink, connected. Without the connection, I can resize all I want and nothing goes wrong. The moment I put the connection in and I resize, then it will crash with this assertion.

jeongcho, 2020-10-14:
I have same probleme.
Does anyone know a solution to this?

segdy, 2020-10-14:
Sorry, I just gave up on Windows and moved to Linux completely.

dkozel, 2021-06-05:
Just reproduced with the RadioConda based 3.9.1.0 release. For me it triggered with 100% reliability when I resized the GRC window across a monitor boundary.

dkozel, 2021-06-05:
Ah! Thanks @MavEtJu for the detailed report! Yes, can also reproduce that the crash does not occur unless there is a connection between blocks.

hllhll, 2021-07-03:
I'm using a multi-screen setup, just installed the latest independent windows installation of 3.8.
It crashes all times when resizing, however, if I close all opened graphs and create a blank one, I can resize and then load my file

paul-k-young, 2022-10-24:
I'm also experiencing similar crashes, however I don't have a multi-display.  The problem appears to be resizing the window in the bottom quarter vertical area of my screen.  I'm running 4k, so that would be about 1.6k-2k pixel range.

Here's a minimum example to reproduce the crash:

1. Open GRC
2. File > New > QT GUI
3. Add a _Null Source_ block
3. Add a _Null Sink_ block
5. Resize the window - Works fine anywhere
6. Connect the source and sink
7. Resize the window - Crashes when the bottom edge is in the lower 3/4 of the screen.  Works otherwise.

Some additional notes:
- Windows 11 and GRC 3.10.4.0
- Display is 4k (3840x2160) @ 120 Hz
- The horizontal size of the window does not seem to affect the behavior.
- Only the displayed graph affects the crash behavior (other tabs have no impact, even if they are connected from source to sink).

mkoepnick, 2022-11-07:
I also observe this problem.  
Windows 10 (10.0.19044)
GRC 3.10.4.0

ryanvolz, 2022-11-10:
I think I have a fix for this in #6352, and it also appears to be highly related to [crashes observed on macOS with the Quartz backend](https://github.com/gnuradio/gnuradio/issues/2726). Since the fix is only in Python code, you can figure out where that code is installed and make the same changes manually to test it out.

paul-k-young, 2022-11-10:
Fixed it for me.  Thank you!

closed 2022-11-17 by mormj