gnuradio/gnuradio #2940: gr3.8 | Stream mux | Dispatches items from input N to the output even if input N+1 does not have items anymore

fmagno, 2019-11-30:
Currently the 

May be arguable whether this is a bug or a feature. IMO, `stream mux` may be more useful if it stops producing output items when these two conditions are simultaneously met:
1. The output produced Nm items relative to the last input stream
2. Any one of the input streams no longer have available items to be sent to the output

Example: consider the following FG.
[image]

- With the current implementation, the content of the output file `stream_mux.dat` is: `0x01 0x01 0x01 0x01`.
- With the above proposed change the content of the file would be: `0x01 0x01 0x01`

mormj, 2019-12-04:
I can see the merits of the suggestions, but I could also see where the naiive implementation of stream mux could be desirable.  For instance if I'm trying to generate a fixed size output file, and bursty input, I wouldn't want stream mux to stop producing outputs if there is not a fixed multiple.  Seems like changing the behavior could break things.

fmagno, 2019-12-07:
I see your point. That makes sense. 👍

[closing the issue]

closed 2019-12-07 by fmagno