gnuradio/gnuradio #2941: undefined symbol: _ZN3uhd11time_spec_t15get_system_timeEv

kewenkang, 2019-12-01:
How to deal with the error below:
ImportError: /usr/lib/python2.7/dist-packages/gnuradio/uhd/_uhd_swig.x86_64-linux-gnu.so: undefined symbol: _ZN3uhd11time_spec_t15get_system_timeEv

michaelld, 2019-12-02:
Guessing you updated UHD without then rebuilding GR ... like 3.13 to 3.14 or the like. With UHD updates, you often have to rebuild GR too, since UHD will have changed enough that the new ABI / library isn't swappable with the prior version.

marcusmueller, 2020-01-06:
Yep, classical problem of conflicting UHD versions, not a GR bug.

closed 2020-01-06 by marcusmueller

Remanm, 2020-07-08:
Hi, I want to know if you have overcome this challenge, I account the same question and dont know what to do

mcan06, 2021-11-23:
Have you solved this problem, could you share with me?

marcusmueller, 2021-11-23:
As said, you've got conflicting UHD versions installed. Remove all but one installations of UHD. 

Also, do not post support questions or reply to these on the github bug tracker. Locking this conversation!