gnuradio/gnuradio #2942: Default "myootConfig.cmake created by modtool searches for non-existent pkg-config file

smunaut, 2019-12-01:
The logic in there includes :

```
INCLUDE(FindPkgConfig)
PKG_CHECK_MODULES(PC_MYOOT myoot)
```

but modtool generates nothing that would create a pkg-config `.pc` file anywhere so this whole logic is unecessary and confusing when trying to understand how this thing works.

smunaut, 2019-12-01:
Is that file even needed at all ? I mean cmake will create a myootConfig.cmake by default automatically and if you remove the useless .pc logic from the manual one, I don't really see what's left of interest ?

marcusmueller, 2024-01-08:
to figure out whether we still need that, we should probably test whether we can make a new module, add a C++ block, build and install the whole shabang, then use the resulting cmake config file in another cmake project.

marcusmueller, 2024-05-05:
Obsoleted by the move towards modern CMake

closed 2024-05-05 by marcusmueller