gnuradio/gnuradio #2943: Generate Error: (TypeError("unsupported operand type(s) for %: 'NoneType' and 'str'"),

luto65, 2019-12-02:
The attached flowgraph when is loaded in GNURadio-6dd3336 for OSX 
[myfm_grc.txt](https://github.com/gnuradio/gnuradio/files/3911313/myfm_grc.txt)
does not generate a .py
Any advice is appreciated.

velichkov, 2019-12-02:
Hi @luto65,

In the `tuner`, `rfgain` and `volume` blocks the `GUI Hint` field is set to `GUI Slider` that is not a valid value. Set it to an empty string or array of four numbers.

See also https://wiki.gnuradio.org/index.php/GUI_Hint and #2287

dkozel, 2019-12-06:
Hi @luto65 I hope Vasil's advice solves your issue. If it does not please post to the discuss-gnuradio mailing list or the chat channel.
https://wiki.gnuradio.org/index.php/MailingLists
https://wiki.gnuradio.org/index.php/IRC

closed 2019-12-06 by dkozel