gnuradio/gnuradio #2944: cmake: Python byte-compilation always uses legacy format filenames

cjmayo, 2019-12-03:
e.g.:
```
/usr/lib/python3.7/site-packages/gnuradio/fft/fft_swig.py
/usr/lib/python3.7/site-packages/gnuradio/fft/fft_swig.pyc
/usr/lib/python3.7/site-packages/gnuradio/fft/fft_swig.pyo
```

I think on >= Python 3.2
https://docs.python.org/3/whatsnew/3.2.html#pep-3147-pyc-repository-directories
these compiled files will be ignored because the source file exists:
https://www.python.org/dev/peps/pep-3147/#case-4-legacy-pyc-files-and-source-less-imports

For packaging on Linux it could even be easier if they weren't compiled via cmake.

skoslowski, 2019-12-04:
The current helper
https://github.com/gnuradio/gnuradio/blob/6414c7c642ab79aa2e96d2e3b394856470cc1444/cmake/Modules/GrPython.cmake#L340
should probably be replaced by https://docs.python.org/3/library/compileall.html

cjmayo, 2019-12-04:
I did try modifying the existing helper, removing `cfile=gen,` the complication being that cmake is constructing filenames for the compiled files, and using them to create specific targets.

marcusmueller, 2024-01-09:
@cjmayo thanks for being persistent on this!

closed 2024-01-16 by willcode