gnuradio/gnuradio #2945: Issue with thrift and maint-3.7 gnu build

mwk088, 2019-12-04:
OS:  CentOS 8
GNU Rev:  maint-3.7
Thrift Rev:  0.10.0
Cmake Rev:  3.11.4

During the make build (from source), GNURADIO finds thrift and enables control port, which is what I want, but during the make process I get the error below.

Any suggestions?

Thank you

Mark

#40 343.6 [ 34%] Built target pygen_gr_vocoder_swig_79184
#40 343.7 /bin/ld: cannot find -lthrift
#40 343.7 collect2: error: ld returned 1 exit status
#40 343.7 make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:1171: gnuradio-runtime/lib/libgnuradio-runtime-3.7.13.5.so.0.0.0] Error 1
#40 343.7 make[1]: *** [CMakeFiles/Makefile2:690: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all] Error 2
#40 343.7 make: *** [Makefile:163: all] Error 2
#40 ERROR: executor failed running [/bin/sh -c source ./install_gnuradio.sh]: exit code: 2

michaelld, 2019-12-04:
Can you do `make VERBOSE=ON` and report back the link command that's generating this error? I'm guessing the issue is that Thrift is installed outside the `CMAKE_INSTALL_PREFIX` and thus we need to provide CMake with some library linkage info.

mwk088, 2019-12-04:
Here is the output of the error.  Let me know if you need more.

Thank you

MArk

```[ 38%] Linking CXX shared library libgnuradio-runtime-3.7.13.5.so
#40 353.3 cd /opt/truearrow/6.2/src/gnuradio/build/gnuradio-runtime/lib && /usr/lib64/python2.7/site-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/gnuradio-runtime.dir/link.txt --verbose=ON
#40 353.3 /usr/bin/c++ -fPIC  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG  -shared -Wl,-soname,libgnuradio-runtime-3.7.13.5.so.0.0.0 -o libgnuradio-runtime-3.7.13.5.so.0.0.0 CMakeFiles/gnuradio-runtime.dir/constants.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter_msgq.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_producer.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_body_wrapper.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_group.cc.o CMakeFiles/gnuradio-runtime.dir/math/fast_atan2f.cc.o CMakeFiles/gnuradio-runtime.dir/math/fxpt.cc.o CMakeFiles/gnuradio-runtime.dir/math/random.cc.o CMakeFiles/gnuradio-runtime.dir/math/sincos.cc.o CMakeFiles/gnuradio-runtime.dir/basic_block.cc.o CMakeFiles/gnuradio-runtime.dir/block.cc.o CMakeFiles/gnuradio-runtime.dir/block_detail.cc.o CMakeFiles/gnuradio-runtime.dir/block_executor.cc.o CMakeFiles/gnuradio-runtime.dir/block_gateway_impl.cc.o CMakeFiles/gnuradio-runtime.dir/block_registry.cc.o CMakeFiles/gnuradio-runtime.dir/buffer.cc.o CMakeFiles/gnuradio-runtime.dir/circular_file.cc.o CMakeFiles/gnuradio-runtime.dir/complex_vec_test.cc.o CMakeFiles/gnuradio-runtime.dir/feval.cc.o CMakeFiles/gnuradio-runtime.dir/flat_flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2_detail.cc.o CMakeFiles/gnuradio-runtime.dir/high_res_timer.cc.o CMakeFiles/gnuradio-runtime.dir/io_signature.cc.o CMakeFiles/gnuradio-runtime.dir/local_sighandler.cc.o CMakeFiles/gnuradio-runtime.dir/logger.cc.o CMakeFiles/gnuradio-runtime.dir/malloc16.c.o CMakeFiles/gnuradio-runtime.dir/message.cc.o CMakeFiles/gnuradio-runtime.dir/misc.cc.o CMakeFiles/gnuradio-runtime.dir/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/msg_handler.cc.o CMakeFiles/gnuradio-runtime.dir/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/pagesize.cc.o CMakeFiles/gnuradio-runtime.dir/prefs.cc.o CMakeFiles/gnuradio-runtime.dir/realtime.cc.o CMakeFiles/gnuradio-runtime.dir/realtime_impl.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_sts.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_tpb.cc.o CMakeFiles/gnuradio-runtime.dir/single_threaded_scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/sptr_magic.cc.o CMakeFiles/gnuradio-runtime.dir/sync_block.cc.o CMakeFiles/gnuradio-runtime.dir/sync_decimator.cc.o CMakeFiles/gnuradio-runtime.dir/sync_interpolator.cc.o CMakeFiles/gnuradio-runtime.dir/sys_paths.cc.o CMakeFiles/gnuradio-runtime.dir/tagged_stream_block.cc.o CMakeFiles/gnuradio-runtime.dir/test.cc.o CMakeFiles/gnuradio-runtime.dir/top_block.cc.o CMakeFiles/gnuradio-runtime.dir/top_block_impl.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_detail.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_thread_body.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_createfilemapping.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_shm_open.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_tmpfile.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_prefs.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_sysv_shm.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcmanager.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcserver_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcserver_booter_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/thrift_application_base.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/gnuradio_types.cpp.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/gnuradio_constants.cpp.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/ControlPort.cpp.o -Wl,-rpath,/opt/truearrow/6.2/src/gnuradio/build/gnuradio-runtime/lib/pmt:/opt/truearrow/6.2/src/gnuradio/build/volk/lib: -lthrift pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0 ../../volk/lib/libvolk.so.1.4 -lboost_date_time -lboost_program_options -lboost_filesystem -lboost_system -lboost_regex -lboost_thread -lpthread -lboost_chrono -lboost_atomic -llog4cpp -lrt -ldl 
#40 353.5 /bin/ld: cannot find -lthrift
#40 353.5 collect2: error: ld returned 1 exit status
#40 353.5 make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:1171: gnuradio-runtime/lib/libgnuradio-runtime-3.7.13.5.so.0.0.0] Error 1```

mwk088, 2019-12-04:
I even get a failure if I don't previously install apache thrift, thus GNU isn't looking for it.

```#40 206.4 cd /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib && /usr/bin/cmake -E cmake_symlink_library libgnuradio-blocks-3.7.13.5.so.0.0.0 libgnuradio-blocks-3.7.13.5.so.0.0.0 libgnuradio-blocks-3.7.13.5.so
#40 206.4 cd /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib && /usr/bin/cmake -E create_symlink libgnuradio-blocks-3.7.13.5.so.0.0.0 /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib/libgnuradio-blocks.so
#40 206.4 cd /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib && /usr/bin/cmake -E create_symlink libgnuradio-blocks-3.7.13.5.so.0.0.0 /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib/libgnuradio-blocks-3.7.13.5.so.0
#40 206.5 cd /opt/truearrow/6.2/src/gnuradio/build/gr-blocks/lib && /usr/bin/cmake -E touch libgnuradio-blocks-3.7.13.5.so.0.0.0
#40 206.5 make[2]: Leaving directory '/opt/truearrow/6.2/src/gnuradio/build'
#40 206.5 [ 68%] Built target gnuradio-blocks
#40 206.5 make[1]: Leaving directory '/opt/truearrow/6.2/src/gnuradio/build'
#40 206.5 make: *** [Makefile:163: all] Error 2
#40 ERROR: executor failed running [/bin/sh -c source ./install_gnuradio.sh]: exit code: 2
```

michaelld, 2019-12-04:
did you clear out the build directory (or use a different one) between builds? pretty sure if you change dependencies you can't reuse the build directory between them.

michaelld, 2019-12-04:
cmake isn't smart enough!

michaelld, 2019-12-04:
so yeah the issue is that the Thrift library directory isn't being added to that set of paths. should be pretty straight forward to correct at least!

michaelld, 2019-12-04:
or ... looks like this is the case; I haven't checked the code yet.

mwk088, 2019-12-04:
I am using a docker file, and thus any changes to script it will rebuild the build folder.

How do I add it to the path?  I checked that the Thrift library is in the path of the ld_library_path.

michaelld, 2019-12-04:
reading through the code ... gimme a few minutes to see if I can figure out what's up ...

michaelld, 2019-12-04:
hmm ... I'm not sure the correct variable is being set in `find_package(Thrift)` ... need to fix up GR's find thrift too ... sigh ...

michaelld, 2019-12-04:
this is a little involved ... will post a patch here for you to try once I have it ...

michaelld, 2019-12-04:
at the top-level "build" directory there will be a file `CMakeCache.txt` ... can you attach it here?

mwk088, 2019-12-04:
[CMakeCache.txt](https://github.com/gnuradio/gnuradio/files/3923220/CMakeCache.txt)

Here ya go.

michaelld, 2019-12-04:
Thx. Interesting ... mine also provides full real path to the library:
```
THRIFT_LIBRARIES:FILEPATH=/opt/truearrow/6.2/lib/libthrift.so
```
This variable is added to the GR libraries:
```
list(APPEND gnuradio_runtime_libs
  ${THRIFT_LIBRARIES}
)
```
which are then used for linking GR-runtime:
```
target_link_libraries(gnuradio-runtime ${gnuradio_runtime_libs})
```

The old-school way to do this would be to add `link_libraries(${THRIFT_LIBRARY_DIRS})` ... which currently doesn't exist because FindThrift doesn't set it. Modern CMake says this about `link_libraries` < https://cmake.org/cmake/help/latest/command/link_directories.html >
{{{
Note This command is rarely necessary and should be avoided where there are other choices. Prefer to pass full absolute paths to libraries where possible, since this ensures the correct library will always be linked. The find_library() command provides the full path, which can generally be used directly in calls to target_link_libraries(). 
}}}

So ... I'm not sure what's going on now! Clearly having the library in a non-standard location is not working ... confusing CMake ... the link command contains "-lthrift" rather than just referencing the actual file including the path to it ... thought admittedly it's doing the same for all of the libraries being linked to GR-runtime ... hmmmm ....

michaelld, 2019-12-04:
what version of CMake are you using?

michaelld, 2019-12-04:
also: are you trying to build GR 3.7.13.5, or GR git branch maint-3.7? or some other GR37 variety?

michaelld, 2019-12-04:
Interesting: My cmake uses the whole path+library for linking. I wonder if this is a cmake version thing or that the libraries are in `${CMAKE_INSTALL_PATH}/lib` ... hmmm ... thinking about how to test ...

mwk088, 2019-12-04:
i am using Cmake Rev: 3.11.4 and cloning in gnuradio branch maint-3.7.

I can change the Cmake rev....it just needs to be 3.5.1 or higher because of UHD-3.15-LTS, which requires it.

michaelld, 2019-12-04:
Can you try the recent 3.16.0 release? If that works, then we know it's a CMake change.

mwk088, 2019-12-04:
It failed again, with the same Thrift error.  Am I not installing Thrift correctly?

mwk088, 2019-12-04:
Latest CMakeCache.

[CMakeCache.txt](https://github.com/gnuradio/gnuradio/files/3923488/CMakeCache.txt)

michaelld, 2019-12-04:
That file looks a -lot- better! Can you cut-and-paste the linker error for this newer CMake?

mwk088, 2019-12-04:
```#48 320.2 [ 38%] Linking CXX shared library libgnuradio-runtime-3.7.13.5.so
#48 320.2 cd /opt/truearrow/6.2/src/gnuradio/build/gnuradio-runtime/lib && /opt/cmake/bin/cmake -E cmake_link_script CMakeFiles/gnuradio-runtime.dir/link.txt --verbose=ON
#48 320.2 /usr/bin/c++ -fPIC  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -O3 -DNDEBUG  -shared -Wl,-soname,libgnuradio-runtime-3.7.13.5.so.0.0.0 -o libgnuradio-runtime-3.7.13.5.so.0.0.0 CMakeFiles/gnuradio-runtime.dir/constants.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter_msgq.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_producer.cc.o CMakeFiles/gnuradio-runtime.dir/messages/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_body_wrapper.cc.o CMakeFiles/gnuradio-runtime.dir/thread/thread_group.cc.o CMakeFiles/gnuradio-runtime.dir/math/fast_atan2f.cc.o CMakeFiles/gnuradio-runtime.dir/math/fxpt.cc.o CMakeFiles/gnuradio-runtime.dir/math/random.cc.o CMakeFiles/gnuradio-runtime.dir/math/sincos.cc.o CMakeFiles/gnuradio-runtime.dir/basic_block.cc.o CMakeFiles/gnuradio-runtime.dir/block.cc.o CMakeFiles/gnuradio-runtime.dir/block_detail.cc.o CMakeFiles/gnuradio-runtime.dir/block_executor.cc.o CMakeFiles/gnuradio-runtime.dir/block_gateway_impl.cc.o CMakeFiles/gnuradio-runtime.dir/block_registry.cc.o CMakeFiles/gnuradio-runtime.dir/buffer.cc.o CMakeFiles/gnuradio-runtime.dir/circular_file.cc.o CMakeFiles/gnuradio-runtime.dir/complex_vec_test.cc.o CMakeFiles/gnuradio-runtime.dir/feval.cc.o CMakeFiles/gnuradio-runtime.dir/flat_flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/flowgraph.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2.cc.o CMakeFiles/gnuradio-runtime.dir/hier_block2_detail.cc.o CMakeFiles/gnuradio-runtime.dir/high_res_timer.cc.o CMakeFiles/gnuradio-runtime.dir/io_signature.cc.o CMakeFiles/gnuradio-runtime.dir/local_sighandler.cc.o CMakeFiles/gnuradio-runtime.dir/logger.cc.o CMakeFiles/gnuradio-runtime.dir/malloc16.c.o CMakeFiles/gnuradio-runtime.dir/message.cc.o CMakeFiles/gnuradio-runtime.dir/misc.cc.o CMakeFiles/gnuradio-runtime.dir/msg_accepter.cc.o CMakeFiles/gnuradio-runtime.dir/msg_handler.cc.o CMakeFiles/gnuradio-runtime.dir/msg_queue.cc.o CMakeFiles/gnuradio-runtime.dir/pagesize.cc.o CMakeFiles/gnuradio-runtime.dir/prefs.cc.o CMakeFiles/gnuradio-runtime.dir/realtime.cc.o CMakeFiles/gnuradio-runtime.dir/realtime_impl.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_sts.cc.o CMakeFiles/gnuradio-runtime.dir/scheduler_tpb.cc.o CMakeFiles/gnuradio-runtime.dir/single_threaded_scheduler.cc.o CMakeFiles/gnuradio-runtime.dir/sptr_magic.cc.o CMakeFiles/gnuradio-runtime.dir/sync_block.cc.o CMakeFiles/gnuradio-runtime.dir/sync_decimator.cc.o CMakeFiles/gnuradio-runtime.dir/sync_interpolator.cc.o CMakeFiles/gnuradio-runtime.dir/sys_paths.cc.o CMakeFiles/gnuradio-runtime.dir/tagged_stream_block.cc.o CMakeFiles/gnuradio-runtime.dir/test.cc.o CMakeFiles/gnuradio-runtime.dir/top_block.cc.o CMakeFiles/gnuradio-runtime.dir/top_block_impl.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_detail.cc.o CMakeFiles/gnuradio-runtime.dir/tpb_thread_body.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_createfilemapping.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_shm_open.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_tmpfile.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_prefs.cc.o CMakeFiles/gnuradio-runtime.dir/vmcircbuf_sysv_shm.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcmanager.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcserver_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcserver_booter_thrift.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/thrift_application_base.cc.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/gnuradio_types.cpp.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/gnuradio_constants.cpp.o CMakeFiles/gnuradio-runtime.dir/controlport/thrift/ControlPort.cpp.o  -Wl,-rpath,/opt/truearrow/6.2/src/gnuradio/build/gnuradio-runtime/lib/pmt:/opt/truearrow/6.2/src/gnuradio/build/volk/lib: -lthrift pmt/libgnuradio-pmt-3.7.13.5.so.0.0.0 ../../volk/lib/libvolk.so.1.4 -lboost_date_time -lboost_program_options -lboost_filesystem -lboost_system -lboost_regex -lboost_thread -lpthread -lboost_chrono -lboost_atomic -llog4cpp -lrt -ldl 
#48 320.5 /bin/ld: cannot find -lthrift
#48 320.5 collect2: error: ld returned 1 exit status
#48 320.5 make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:1171: gnuradio-runtime/lib/libgnuradio-runtime-3.7.13.5.so.0.0.0] Error 1```

michaelld, 2019-12-04:
same same ... grrrr ... should be different!

In the build log, does it verify the cmake version is as desired?

Did you clean out the build directory and/or re-do the build entirely from start using the new cmake version?

michaelld, 2019-12-04:
OK so in the cache the cmake version is 3.16.0 ... so, that's good!

michaelld, 2019-12-04:
guessing since the cache is updated, you started with a fresh build too

mwk088, 2019-12-04:
yep, I am still at a loss as to why the library is not being linked.

michaelld, 2019-12-04:
Can you try a quickie patch? Edit the file "gnuradio-runtime/lib/CMakeCache.txt" ... search for "SHARED LIB BUILD". Just before that, add `link_directories("/opt/truearrow/6.2/lib/")`. Try the build again (don't need to clear the build; just restart it) & see if that works; it should now.

mwk088, 2019-12-04:
I would, but I do not see any CMakeCache.txt file in the directory.

michaelld, 2019-12-04:
arg ... `CMakeLists.txt` ... sorry!

michaelld, 2019-12-04:
So: gnuradio-runtime/lib/CMakeLists.txt ...

mwk088, 2019-12-04:
No problem.  

This makes no sense....I modified the file you suggested and reran make, but instead cmake ran as well...the make then ran and same failure.  I went back into the gnuradio-runtime\lib folder to find and modify the CMakelists.txt file, but it is nowhere to be found  What the heck?

michaelld, 2019-12-04:
I do expect cmake to re-run with that change, but I would expect the file "gnuradio-runtime/lib/CMakeLists.txt" to still exist afterwards!

mwk088, 2019-12-04:
Ok, so I then reran cmake then make then make install.....and everything worked.  Doesn't make sense to me.  All of this was done manually inside my docker container opposed to my Dockerfile.

michaelld, 2019-12-04:
Meaning that it now worked in the docker container, but with the "quickie patch" I described ... or, without it?

mwk088, 2019-12-05:
without the quickie patch...which makes no sense to me as to why it won't work in my dockerfile....but will work manually.  I am executing the same commands.

michaelld, 2019-12-05:
OK, so sanity prevails at least with respect to normal builds!

In your dockerfile, even adding in the quickie patch doesn't do the trick, I take it? Now -that's- strange!

mwk088, 2019-12-05:
It is very strange.  Now, for another approach in a different dev environment I decided to simplify my docker build greatly, relying on pybombs more to take care of all the packages minus the ones I need to install manually.  I will let you know how this more methodical approach works out.

mwk088, 2019-12-05:
So, I am trying to have pybombs install everything and I running the command:
pybombs -v -v prefix init ${PB_PREFIXALIAS} -R ${GR_VERSION}
my GR_VERSION is gnuradio-stable
and I keep getting this error:
PyBOMBS.PackageManager - DEBUG - Package python-click-plugins is not installable.
PyBOMBS.install_manager - ERROR - Package has no install method: python-click-plugins
PyBOMBS - DEBUG - Unresolved install path.

michaelld, 2019-12-05:
Can't help you with PyBOMBS ... never used it!

mwk088, 2019-12-05:
Ok, thanks for the feedback.

michaelld, 2019-12-06:
@mwk088 FYI I'm closing this issue, as it has been resolved, well, enough.

closed 2019-12-06 by michaelld