gnuradio/gnuradio #2949: OOT block with CUDA

srinadhpotnuru, 2019-12-05:
I have a question regarding linking external libraries in my own OOT block with external libraries or cuda(.cu) files. Can someone guide me with editing CMakeLists.txt to bring in functionalities of GPU as I am facing difficulties in bridging the two. Hope I can get a reply soon.

Thanks,
Srinath

michaelld, 2019-12-05:
@srinadhpotnuru Please post some files and/or a link to a public repo so we can see the state of your OOT CMake. It's pretty much impossible to provide any specific information based on your query. I believe there is at least 1 GR OOT module that uses CUDA ... so some well-targeted internet searching might turn up what you're looking for.

michaelld, 2019-12-05:
This is not an issue with GR, so I'm closing this. Posting to the GR discussion email list would be the way to go here.

closed 2019-12-05 by michaelld