gnuradio/gnuradio #2950: GRC does not connect() blocks with no ports even though GR supports this

kpreid, 2019-12-06:
Steps to reproduce:

1. Define a hierarchical block which has no input or output ports, but which has blocks that do some detectable work inside it.
2. Define a top block, in GNU Radio Companion, which contains this hierarchical block.
3. Run the generated top block; observe that nothing happens.

The generated code instantiates the hier_block (e.g. `self.hier_block_no_ports_0 = hier_block_no_ports()`) but does not connect it (e.g. `self.connect(self.hier_block_no_ports_0)`, using the 1-argument form of the `connect` method), so the blocks within the hier block do not do anything. If I manually add the `connect` call then the self-contained graph within the hier block does execute.

Thus, GR supports this case but GRC does not. I propose that GRC be modified so that, **if a block in a top/hier block has no ports, then a 1-argument `connect` call will be generated for it.**

Compatibility considerations: As far as I know, this should not affect the behavior of existing correct programs; however, if for some reason a block is declared to GRC to have no ports but actually does have ports, this will cause a new runtime error (whereas the block would previously have been instantiated but otherwise ignored).

dkozel, 2019-12-06:
Thanks for filing the issue. This seems like a clear and hopefully reasonably straightforward improvement to GRC!

haakov, 2020-01-03:
I can take care of this. Note to self: Don't `connect()` blocks that don't have a make template

haakov, 2020-01-04:
@kpreid I've added a PR: https://github.com/gnuradio/gnuradio/pull/3007

Can you please confirm that it fixes the bug?

kpreid, 2020-01-05:
@haakov I attempted, but I'm currently having trouble building GR from source. Sorry. The code looks reasonable to me.

closed 2021-02-11 by mbr0wn