gnuradio/gnuradio #2953: Blocks, Properties dialog too small in size

Phil67ago, 2019-12-09:
It is often too small. The drawback for me is that I always try to scroll with the wheel button.
What often happens is nothing, no scrolling. What I did was changing some drop down box under the cursor.
Annoying because to get back I have to Cancel and open the Dialog again.
This is on Fedora 30.

I have tested a patch against maint-3.8
[gnuradio-prop-dialog-size_maint-3.8.patch.txt](https://github.com/gnuradio/gnuradio/files/3938723/gnuradio-prop-dialog-size_maint-3.8.patch.txt)

Very dirty, someone with better QT knowledge should do it better.

dkozel, 2019-12-09:
@Phil67ago Thanks again for the report and patch. I'm not sure I understand the issue, can you add some screenshots?

Phil67ago, 2019-12-09:
This is from 3.7.13.5
[image]

And this is 3.8.0.0 with patch
[image]

Above is as they are displayed without resizing the Dialog frame.

If I scroll the mouse wheel above 'Window Type' I will change 'Backman-harris' to something else!

dkozel, 2019-12-10:
I see! Yes, this seems like a reasonable change. We'll get a review of the implementation and check what happens on blocks with lots of properties.

devnulling, 2020-01-04:
Testing is needed for this patch. It may cause issues when the window size is larger than the screen, for example with a UHD source/sink that has 32 channels.

devnulling, 2020-01-05:
@Phil67ago I'm going to close this issue as this change will produce undesired behavior when used with a properties dialog that has more options than fit within the screen, for example with a USRP source that has 20 channels, it'll force the top/bottom off screen.

I would like to note though that we are working on porting GRC to use QT, so it will likely have better default behavior with that switch.

closed 2020-01-05 by devnulling