gnuradio/gnuradio #2954: Cannot Run Gnuradio-Companion

bencampionmax, 2019-12-10:
Hello,
I ran this command: 
pybombs run gnuradio-companion
Output:
[INFO] Prefix Python version is: 3.6.8
[INFO] PyBOMBS Version 2.3.4a0
Gtk-Message: 17:31:14.265: GtkDialog mapped without a transient parent. This is discouraged.

Additionally, a ModuleNotFoundError window with the following text:

Cannot import gnuradio.
Is the model path environment variable set correctly?
All OS: PYTHONPATH

Is the library path environment variable set correctly?
Linux: LD_LIBRARY_PATH

velichkov, 2019-12-10:
Hi @bencampionmax,

What happened with the [failing  `qa_qtgui` test](https://github.com/gnuradio/gnuradio/issues/2506#issuecomment-563501885), did you solve the problem and how?

> Additionally, a ModuleNotFoundError window with the following text:
>
> Cannot import gnuradio.
> Is the model path environment variable set correctly?
> All OS: PYTHONPATH

Provide the full output of `pybombs prefix env | grep python` command.

You could seek help for such problems in the [mailing list](https://wiki.gnuradio.org/index.php/MailingLists) or [IRC/Slack](https://wiki.gnuradio.org/index.php/IRC) channels.

bencampionmax, 2019-12-10:
Hi Vasil,

1. The command that you suggested returned: -bash: xvfb-run: command not
found
2. This is the output:
~/pybombs/gnuradio/build$ pybombs prefix env | grep python
[INFO] Prefix Python version is: 3.6.8
[INFO] PyBOMBS Version 2.3.4a0
PYTHONPATH=/home/ben/prefix/default/lib/python3.6/site-packages:/home/ben/prefix/default/lib/python3.6/dist-packages:/home/ben/prefix/default/lib64/python3.6/site-packages:/home/ben/prefix/default/lib64/python3.6/dist-packages:/home/ben/prefix/default/lib/python3.6/site-packages:/home/ben/prefix/default/lib/python3.6/dist-packages:/home/ben/prefix/default/lib64/python3.6/site-packages:/home/ben/prefix/default/lib64/python3.6/dist-packages:

Thanks,
Ben

velichkov, 2019-12-10:
> 1. The command that you suggested returned: -bash: xvfb-run: command not found

You need to install `xvfb` first. On Debian/Ubuntu you could run `sudo apt-get install xvfb`.
Note that this is needed only if you are running the tests on a headless system (without GUI - X11, Wayland, ...) in a CI/CD for example or on a remote server through ssh.

The gnuradio-companion is a GUI application, you can't really use it under xvfb but you could run the tests using xvfb.

> 2. This is the output: ~/pybombs/gnuradio/build$ pybombs prefix env | grep python [INFO] Prefix Python version is: 3.6.8 [INFO] PyBOMBS Version 2.3.4a0 [...]

Provide the full output of the following commands
```
pybombs config
pybombs inv
find /home/ben/prefix/default/lib* -name gnuradio
cat /home/ben/prefix/default/setup_env.sh
```

bencampionmax, 2019-12-10:
>    1. This is the output: ~/pybombs/gnuradio/build$ pybombs prefix env |

1. After installing xvfb and running the previous command the output is:
UpdateCTestConfiguration  from
:/home/ben/pybombs/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from
:/home/ben/pybombs/gnuradio/build/DartConfiguration.tcl
Test project /home/ben/pybombs/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 347
    Start 347: qa_qtgui

347: Test command: /bin/sh
"/home/ben/pybombs/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
347: Test timeout computed to be: 9.99988e+06
347: Segmentation fault
1/1 Test #347: qa_qtgui .........................***Failed    0.31 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.32 sec

The following tests FAILED:
        347 - qa_qtgui (Failed)
Errors while running CTest

2. *Command:* pybombs config
*Output:*
[INFO] Prefix Python version is: 3.6.8
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Using config file: /home/ben/.pybombs/config.yml
builddocs: ON
  - Build doxygen while compiling packages? options are: ON, OFF
git-cache: /home/ben/.pybombs/gitcache
  - Path to git reference repository (git cache)
elevate_pre_args: ['sudo', '-H']
  - For commands that need elevated privileges, prepend this
makewidth: 48
  - Concurrent make threads [1,2,4,8...]
cmakebuildtype: RelWithDebInfo
  - CMAKE_BUILD_TYPE args to pass to cmake projects, options are: Debug,
Release, RelWithDebInfo, MinSizeRel
packagers: apt,pymod,pip,pkgconfig,cmd
  - Priority of non-source package managers
default_prefix: /home/ben/prefix
  - Default Prefix
keep_builddir:
  - When rebuilding, default to keeping the build directory

*Command:* pybombs inv
*Output:*
[INFO] Prefix Python version is: 3.6.8
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Using config file: /home/ben/.pybombs/config.yml
builddocs: ON
  - Build doxygen while compiling packages? options are: ON, OFF
git-cache: /home/ben/.pybombs/gitcache
  - Path to git reference repository (git cache)
elevate_pre_args: ['sudo', '-H']
  - For commands that need elevated privileges, prepend this
makewidth: 48
  - Concurrent make threads [1,2,4,8...]
cmakebuildtype: RelWithDebInfo
  - CMAKE_BUILD_TYPE args to pass to cmake projects, options are: Debug,
Release, RelWithDebInfo, MinSizeRel
packagers: apt,pymod,pip,pkgconfig,cmd
  - Priority of non-source package managers
default_prefix: /home/ben/prefix
  - Default Prefix
keep_builddir:
  - When rebuilding, default to keeping the build directory
ben@hydradesktop:~/pybombs/gnuradio/build$ ^C
ben@hydradesktop:~/pybombs/gnuradio/build$ pybombs inv
[INFO] Prefix Python version is: 3.6.8
[INFO] PyBOMBS Version 2.3.4a0
Showing package state:

*Command:* find /home/ben/prefix/default/lib* -name gnuradio
*Output:*
find: ‘/home/ben/prefix/default/lib*’: No such file or directory

*Command: *cat /home/ben/prefix/default/setup_env.sh
# WARNING: This file is auto-generated by pybombs, any manual changes to it
may be overwritten!
export PATH="/home/ben/prefix/default/bin:$PATH"
export
PYTHONPATH="/home/ben/prefix/default/lib/python3.6/site-packages:/home/ben/prefix/default/lib/python3.6/dist-packages:/home/ben/prefix/default/lib64/python3.6/site-packages:/home/ben/prefix/default/lib64/python3.6/dist-packages:$PYTHONPATH"
export
LD_LIBRARY_PATH="/home/ben/prefix/default/lib:/home/ben/prefix/default/lib64/:$LD_LIBRARY_PATH"
export
LIBRARY_PATH="/home/ben/prefix/default/lib:/home/ben/prefix/default/lib64/:$LIBRARY_PATH"
export
PKG_CONFIG_PATH="/home/ben/prefix/default/lib/pkgconfig:/home/ben/prefix/default/lib64/pkgconfig:$PKG_CONFIG_PATH"
export PYBOMBS_PREFIX="/home/ben/prefix/default"
# If we're in a Python virtualenv, activate that
if [ -r /home/ben/prefix/default/bin/activate ]; then
    source /home/ben/prefix/default/bin/activate
fi

velichkov, 2019-12-10:
> test 347
>    Start 347: qa_qtgui
>
> 347: Test command: /bin/sh
> "/home/ben/pybombs/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
> 347: Test timeout computed to be: 9.99988e+06
> 347: Segmentation fault
> 1/1 Test #347: qa_qtgui .........................***Failed    0.31 sec

Which Linux distribution do you use and which version?
Did you manually check out gnuradio in `/home/ben/pybombs/gnuradio/` and are manually building it, e.g. without using `pybombs`?

> ben@hydradesktop:~/pybombs/gnuradio/build$ pybombs inv
> [INFO] Prefix Python version is: 3.6.8
> [INFO] PyBOMBS Version 2.3.4a0
> Showing package state:

> *Command:* find /home/ben/prefix/default/lib* -name gnuradio
> *Output:*
> find: ‘/home/ben/prefix/default/lib*’: No such file or directory

The empty output of `pybombs inv` and the `No such file or directory` error indicate that you **have not** installed gnuradio in your prefix.

```
pybombs prefix init /home/ben/prefix/default/ -R gnuradio-default
```
See also https://github.com/gnuradio/pybombs/#to-run-pybombs-quickstart

Have you attempted installing gnuradio from the distribution package repository?
Provide the output of `which gnuradio-companion`

shrin18, 2019-12-11:
> Hello,
> I ran this command:
> pybombs run gnuradio-companion
> Output:
> [INFO] Prefix Python version is: 3.6.8
> [INFO] PyBOMBS Version 2.3.4a0
> Gtk-Message: 17:31:14.265: GtkDialog mapped without a [...]

`export PYTHONPATH=$PYTHONPATH:git fetch`
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib`

You can try these commands. I faced similar problem earlier.

marcusmueller, 2020-04-12:
@shrin18 these make no sense in this context.

marcusmueller, 2020-04-12:
Closing due to no response in four months.

closed 2020-04-12 by marcusmueller

m4ver1cks, 2020-04-13:
II solved the issue adding to the setup_env.sh a new entry in the PYTHONPATH with the real path of the python files (the pybombs do not add it correctly)