gnuradio/gnuradio #2956: Problem when using USRP sink and USRP source in the same flowgraph

YunWen2019, 2019-12-12:
I want to use the received signals from USRP source to control the transmission signals into USRP sink in the same flowgraph of gnuradio-companion, but it doesn't works well. 

The flowgraph is somehow like 

USRP source -> some control blocks and random signal source -> USRP sink  

but the USRP sink doesn't work. It doesn’t transmit at all while flowgraph is running without any error, or just transmits in an intermittent manner (as observed in an external spectrum analyzer). It also outputs lots of “L” in command window thus I can only terminate it by terminating the gnuradio-companion.

When I replace the USRP source with a random source, USRP sink works and I can see the continuous transmit signals in spectrum analyzer. 
In order to emit the effect from unstable control output from received signal, I test to process it through a Or block with a constant source 1. I expect that the USRP sink can transmit in a continuous manner in this case, but it still doesn’t not work.  

I also tried to set USRP source and sink with different clocks (one with PPS and one with PC-clock), or different frequencies, but all failed.
In my flowgraph, the USRP source and USRP sink use TX/RX and RX2 in the same daughter board of X310, respectively.
(There is an attenuator in the loopback part between TX/RX and RX2) 

My OS is Ubuntu 18.04, with gnuradio 3.8, gcc 7.4.0, Python 3.6 and uhd 3.15.0 

Is there anyone can help me? Thanks in advance

Yun

YunWen2019, 2019-12-19:
I'd like to correct my issue that the error outputs are lots of "U" but not "L".
#2491 has same error outputs with mine, I tried its method but still can not solve my problem.

devnulling, 2020-01-04:
@YunWen2019 what sample rate are you running your flowgraph at?

dkozel, 2021-07-10:
I'm going to close this as is almost certainly a USRP/UHD setup issue and has been inactive for a year. If anyone encounters this issue again please let us know.

closed 2021-07-10 by dkozel

rev8051, 2023-09-22:
What is the answer?!!

marcusmueller, 2023-09-22:
@rev8051 your comment is not helpful. If you have a similar problem but one that is actually a GNU Radio issue, please open a new issue report.