gnuradio/gnuradio #2960: tagged_stream_block ignores rate changes for additional tags

jdemel, 2019-12-16:
I use the `fec_tagged_encoder` block because I want my length tags to be updated correctly.

All other tags are lost because their position is not updated according to the relative rate change.

Setting `enable_update_rate(true)` in
```
tagged_stream_block::tagged_stream_block( ... )
    : ...
{
    enable_update_rate(true);
}
```
fixes the problem. I wonder if this is the way to go in this case. Or does that lead to unwanted side-effects?
Or should we reimplement `propagate_tags`?
Or should we check `relative_rate` and only update relative rate if this is required?

michaelld, 2020-01-10:
Can we set `enable_update_rate(true);` in the constructor for `fec_tagged_encoder` with the same effect? If so, that'd be my vote, because my guess is that globally using this setting would impact some blocks that rely on it being `false` by default.

jdemel, 2020-01-11:
Yeah. We can do that in `fec_tagged_encoder` but then we need a couple more blocks to add this. I can do this for my work whenever I come across a block that needs this setting. But that would lead to inconsistency among blocks in the long run.
Can we set it to `true` by default and set it back to `false` for blocks that rely on it? Also, which blocks do rely on it?

marcusmueller, 2024-01-08:
Interesting situation, especially in the face of the question whether we still want to touch TSBs; anyways, will need to reproduce first, unless @jdemel  knows this isn't an issue anymore.