gnuradio/gnuradio #2961: Python3 unable to parse cons-wrapped-cons

MileB, 2019-12-16:
Assume an existing dictionary of meta-data, **meta**, and a vector of complex samples, **out**.

If one tries to pass these bundled up in to a cons, and **meta** contains a cons itself, the receiver (if python) will be unable to parse the embedded cons.

**Somewhat-pseudo C code**:
```C++
meta = dict_add(meta, string_to_symbol('key'), cons(12.34, 56.78));
message_port_pub("out", cons(meta, out));
```
**Somewhat-pseudo Python code**
```python
bits = socket.recv()
pdu = pmt.deserialize_str(bits)
meta = pmt.to_python(pmt.car(pdu)) # Crashes
```

The stack of the crash hints the following:
>  RuntimeError: pmt_length: wrong_type : (12.34 . 56.78)

This appears to be coming from the call 
```python
for i in range(pmt.length(items))
```
of `pmt_to_dict`

MileB, 2019-12-19:
It looks like this behavior extends in to the uhd blocks as well.

uhd_async_msg can't be parsed out because the field **time_spec** falls into this rule.

MileB, 2019-12-20:
Found another hint.

It looks to be because these guys are passing the test for `is_dict` in pmt_to_python
[here](../blob/master/gnuradio-runtime/python/pmt/pmt_to_python.py#L117). If instead they hit the following block, `is_pair`, it looks like they unwrap properly.

MileB, 2019-12-23:
I've figured out what exactly is going wrong here as well as a potential solution.

## The Problem 

[**pmt_to_python**](../blob/master/gnuradio-runtime/python/pmt/pmt_to_python.py#L123) Iterates across all supported types, trying to convert a given PMT to the proper python typing. If a conversion fails (throws `TypeError` or `ValueError`) the loop iterates to the next type, until one sticks or all have been exhausted.

**This works fine in Python2. This is not sufficient in Python3.**

Python3 will not throw a `TypeError` or `ValueError` when trying to convert a **cons** that's been found [here](../blob/master/gnuradio-runtime/python/pmt/pmt_to_python.py#L61). I.e. if you have a cons within a dictionary.

What happens is:
- User calls `pmt.to_python(d)` where d is a dictionary containing a cons.
- `pmt_to_python` correctly resolves to true for [pmt.is_dict](../blob/master/gnuradio-runtime/python/pmt/pmt_to_python.py#L117) 
- `pmt_to_dict` iterates across all key/value pairs. Eventually hits the cons.
- `pmt_to_python` resolves to true for `pmt.is_dict` on the cons. 
- `pmt.length` throws an exception because we're not actually a dictionary.
    - In Python2 this will get caught [here](../blob/master/gnuradio-runtime/python/pmt/pmt_to_python.py#L127) and the code just tries again, interpreting the cons as a list.
   - **In Python3,** a `RuntimeError` bubbles up instead, which isn't handled. This leads to our above error. 

## The Solution
The current approach to try each type is fine. The problem is that we're not throwing the right exception when a wrong type is hit.

The relevant sections for these exceptions are:
- [pmt.h](../blob/master/gnuradio-runtime/include/pmt/pmt.h#L104)
- [pmt.cc](../blob/master/gnuradio-runtime/lib/pmt/pmt.cc#L51)
- [gr_extras.i](../blob/master/gnuradio-runtime/swig/gr_extras.i#L34)

The [Swig Exception Library](http://www.swig.org/Doc3.0/Customization.html#Customization_nn7) has the option to throw a `SWIG_TypeError`, which would resolve to a `TypeError` that the Python loop handles.

I feel like this is a cleaner solution than just adding `RuntimeError` to the exception checking list in `pmt_to_python`, though that would certainly work too.

I've attached a patch file that resolves my issue here. I'm not a big C++ writer, but of their [exceptions](https://en.cppreference.com/w/cpp/error) it seemed like `invalid_argument` made sense. Anything that's more specific than the catch-all of `exception` would work.

[pmt-patch.txt](https://github.com/gnuradio/gnuradio/files/3995522/pmt-patch.txt)

devnulling, 2020-01-23:
@MileB would you want to submit that patch as a PR?

MileB, 2020-01-23:
> @MileB would you want to submit that patch as a PR?

I've created one here: #3037

PR #3037 (2020-02-09): pmt: Fix RuntimeError in pmt_to_python
Python3 now has a RuntimeError that's thrown during the pmt to_python function.
Instead of throwing this error, SWIG has been updated to throw a TypeError.
This allows us to keep the same behavior whereby we iterate over PMT types until the proper conversion is found

Resolves #2961
closed 2020-02-09 by marcusmueller