gnuradio/gnuradio #2962: grc: DeprecationWarning: 'encoding' is ignored and deprecated. It will be removed in Python 3.9

nickoe, 2019-12-17:
I get this error spammed when running grc, on archlinux with python 3.8.0 and gnuradio  3.8.0.0. I could not see any issue or pull request specifically addressing this issue, so I assume it is still an issue on `master` and the `maint-3.8` branch, but I didn't verify this.

```
/usr/lib/python3.8/site-packages/gnuradio/grc/core/utils/extract_docs.py:201: DeprecationWarning: 'encoding' is ignored and deprecated. It will be removed in Python 3.9
  key, cmd, args = json.loads(line.decode('utf-8'), encoding='utf-8')

```

skoslowski, 2019-12-17:
https://github.com/gnuradio/gnuradio/pull/2913

nickoe, 2019-12-17:
@skoslowski thank you. But should't this be cherry picked to maint-3.8 as well?

devnulling, 2020-01-19:
Closing this as it was back ported to maint-3.8 with https://github.com/gnuradio/gnuradio/pull/2994

closed 2020-01-19 by devnulling