gnuradio/gnuradio #2973: Python min version exists, but is not found

kaluzaklein, 2019-12-22:
I'm trying to install gnuradio with pybombs using the Python 2 installation instructions.
My first comment is that I have to install several python 3 libraries (e.g.: python3 -m pip install numpy six mako ) in order to progress further in the installation with fewer errors. This is strange, because I'm installing for Python 2, not Python 3.

The error is that PYTHON_MIN_VER_FOUND is FALSE, when in reality I have Python 2.7.13 which satisfies the minimum version requirement.

Also, I'm not sure how to install Cairo (GI).

Commands I'm using from https://github.com/gnuradio/pybombs :

sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs prefix init ~/tools/gr38 -R gnuradio-default

I'm using very fresh install of linux on a linux VM (pengiun/Debin linux VM on a Pixelbook).
python --version
Python 2.7.13
python3 --version
Python 3.5.3

I did "sudo apt-get install libssl-dev" and installed cmkae 3.16.12.

Here is the pybombs error:
--------------
```
-- Python checking for Cairo (GI) >= 1.0 - not found
-- Python checking for PangoCairo (GI) >= 1.0 - found
-- Python checking for numpy - found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = FALSE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = FALSE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:95 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
```

velichkov, 2019-12-23:
Hi @kaluzaklein,

> I'm trying to install gnuradio with pybombs using the Python 2 installation instructions.
> My first comment is that I have to install several python 3 libraries (e.g.: python3 -m pip install numpy [...]

Try adding `-DPYTHON_EXECUTABLE=/usr/bin/python2` in the `config_opt:` parameter in `~/.pybombs/recipes/gr-recipes/gnuradio.lwr` receipe.

```diff
diff --git a/gnuradio.lwr b/gnuradio.lwr
index 2fc4948..3137588 100644
--- a/gnuradio.lwr
+++ b/gnuradio.lwr
@@ -56,7 +56,7 @@ source: git+https://github.com/gnuradio/gnuradio.git
 gitbranch: maint-3.8
 gitargs: --recursive
 vars: # We explicitly enable some subcomponents to make sure the build fails if they're not working:
-  config_opt: " -DENABLE_DOXYGEN=$builddocs -DENABLE_GR_AUDIO=ON -DENABLE_GR_BLOCKS=ON -DENABLE_GR_DIGITAL=ON -DENABLE_GR_FEC=ON -DENABLE_GR_FFT=ON -DENABLE_GR_FILTER=ON -DENABLE_GR_QTGUI=ON -DENABLE_GR_UHD=ON -DENABLE_PYTHON=ON -DENABLE_VOLK=ON -DENABLE_GRC=ON "
+  config_opt: " -DENABLE_DOXYGEN=$builddocs -DENABLE_GR_AUDIO=ON -DENABLE_GR_BLOCKS=ON -DENABLE_GR_DAIGITAL=ON -DENABLE_GR_FEC=ON -DENABLE_GR_FFT=ON -DENABLE_GR_FILTER=ON -DENABLE_GR_QTGUI=ON -DENABLE_GR_UHD=ON -DENABLE_PYTHON=ON -DENABLE_VOLK=ON -DENABLE_GRC=ON -DPYTHON_EXECUTABLE=/usr/bin/python2"
 inherit: cmake
 configure_static: cmake .. -DCMAKE_BUILD_TYPE=$cmakebuildtype -DCMAKE_INSTALL_PREFIX=$prefix -DENABLE_STATIC_LIBS=True $config_opt
 install: |
```

devnulling, 2020-01-23:
@kaluzaklein  is this still an outstanding issue ? Can this issue be closed?

devnulling, 2020-01-24:
I'm going to close this, please feel free to reopen if it's still an issue.

closed 2020-01-24 by devnulling