gnuradio/gnuradio #2980: no osmocom source in gnuradio-companion

mahescho, 2019-12-25:
I've installed following https://wiki.gnuradio.org/index.php/InstallingGR using the PPA. I've a RTL-SDR stick. Now in gnuradio-companion there is no osmocom or RTL source. I've read that RTL is now part of the osmocom source. In /usr/share/gnuradio/grc/blocks "osmosdr_source.xml" exists.

How to get/find the osmocom source in gnuradio-companion?

velichkov, 2019-12-27:
Hi @mahescho,

> I've installed following https://wiki.gnuradio.org/index.php/InstallingGR using the PPA. 

From which PPA did you install it (as there are [four PPA repositories](https://launchpad.net/~gnuradio))?

> I've a RTL-SDR stick. Now in gnuradio-companion there is no osmocom or RTL source. I've read that RTL is now part of the osmocom source. In /usr/share/gnuradio/grc/blocks "osmosdr_source.xml" exists.
>
> How to get/find the osmocom source in gnuradio-companion?

How did you install `gr-osmosdr`? You need to install it from source and not from the ubuntu's repository, and for gnuradio 3.8 you need to install it from the [gr3.8 branch](https://github.com/osmocom/gr-osmosdr/tree/gr3.8). Also make sure to first uninstall your current version.

The-inside-man, 2019-12-30:
I am having so many issues with gnuradio... tried off the 3.8 branch and will not compile. Getting header issues and such. Tried upgrading my cmake and now have issues with version 3.8 of gnuradio and cmake Version not being compatible. I am using cmake 3.16 with gnuradio 3.8. Anyone else have the same issues. I’ve had no luck whatsoever with getting gr-osmosdr installed as it fails on make.

I cloned the repo for gr-osmosdr and tried from there. On Ubuntu it defaults to cmake 3.10.2. When I switched to use the source from gr3.8 my cmake was out of date and only compatible for 3.12 and above. I upgraded to the 3.16 version as mentioned. This caused a new issue with gnuradio’s .cmake and osmosdr wish a mismatch between 3.7 and 3.9 version... heard of anything like this?

devnulling, 2020-01-04:
Hi @mahescho , @The-inside-man 

Questions such as this are better served either on the mailing list [0], on Slack (register here [1] and login here [2]), or on IRC [3].

[0] - https://wiki.gnuradio.org/index.php/MailingLists
[1] - https://slack.gnuradio.org/
[2] - https://gnuradio.slack.com/
[3] - https://wiki.gnuradio.org/index.php/IRC

closed 2020-01-04 by devnulling

rileyrg, 2021-10-10:
hmmm Oct 2021 and 3.8.2 from Debian Testing has no sources either. I'm literally just starting out on this so thought I should mention it.

dkozel, 2022-02-17:
@alx-xlx it is available via the recommended RadioConda installer which became fully useful since this issue was closed last year.

https://wiki.gnuradio.org/index.php?title=InstallingGR

jamland, 2022-11-16:
was able to successfully install it from here by steps specified in README
https://github.com/osmocom/gr-osmosdr