gnuradio/gnuradio #2982: grc swallows true error message of `firdes.low_pass()` in

ThomasHabets, 2019-12-25:
## Steps to reproduce

1. Open GRC
1. Create a `frequency xlating fft filter`
1. Double-click on it to open properties
1. Set taps to `firdes.low_pass(1,1,-1,1)` (negative cutoff is invalid)

## Expected behaviour

Error seen below the properties to be the same seen when creating it in python:

```python
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gnuradio.filter import firdes
>>> firdes.low_pass(1,1,-1,1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "[…]/opt/gnuradio-master/lib/python3.7/dist-packages/gnuradio/filter/filter_swig.py", line 143, in low_pass
    return _filter_swig.firdes_low_pass(*args, **kwargs)
RuntimeError: firdes check failed: 0 < fa <= sampling_freq / 2
>>> 
```

## Actual behavior

Errors shown is:

```
Param - Taps(taps):
	Expression None is invalid for type'complex_vector'.
```

# Versions

* GNU Radio HEAD (a402fc432ec2019e5bfc70c3f78beef83b450f01)
* Debian linux 10.2
* Python 3.7.3

mormj, 2020-01-08:
This seems to be a persistent problem, and an improvement possibility for GRC to propagate errors from methods called in variables instead of just returning None

Related: #2832 
GRC has asserts but they are not often used in practice as they are not easy to implement

marcusmueller, 2024-01-08:
Fixed, as things now fail with:

```
Param - Taps(taps):
	Value "firdes.low_pass(1,1,-1,)" cannot be evaluated:
	low_pass(): incompatible function arguments. The following argument types are supported:
	    1. (gain: float, sampling_freq: float, cutoff_freq: float, transition_width: float, window: gnuradio.fft.fft_python.window.win_type = <win_type.WIN_HAMMING: 0>, param: float = 6.76) -> List[float]
	
	Invoked with: 1, 1, -1

Param - Taps(taps):
	Expression None is invalid for type'complex_vector'.
```

Test fg:
https://gist.github.com/marcusmueller/9a3ceff70fd1f23c763c8a427f28d778
[image]

closed 2024-01-08 by marcusmueller