gnuradio/gnuradio #2985: "Symbol Sync" block can't be added to GRC flowgraph

RedStone002, 2019-12-30:
Dear all,
I did a fresh install of GNURadio v3.8.0.0/v1.7 BETA for Windows from the link in GNURadio wiki.
It works almost perfectly, but I noticed one issue:
I can't add the Core-Synchronizers->Symbol Sync to my GRC flowgraph.
When I double click the "Symbol Sync" nothing happens. No error message. No text in the output view. No nothing.
Adding the other synchronizers works fine, just this can't be added to the flowgraph.

Any ideas how to diagnose the problem, or to fix it?

RedStone002, 2019-12-30:
Poking a little bit around, I think this happens when the text
Mueller and M?ller
should be displayed in the drop-down list. The correc text would be "Mueller and Müller" with an the german Umlaut ü.

RedStone002, 2019-12-30:
I found the offending string in the file
`gnuradio\share\gnuradio\grc\blocks\digital_symbol_sync_xx.block.yml`

In line 22 it reads:
> option_labels: ["Mueller and M\xFCller", "Modified Mueller and M\xFCller",
If I change it to this, the block can be inserted without eror:
>option_labels: ["Mueller and Mueller", "Modified Mueller and Mueller", ... ...
 
The underlying issue is a little deeper. Is the YML file supposed be ASCII or UTF-8?
Is the data read byte-wise or as strings? Is the encoding handled consistently?
Further problems ahead: Python 2.7 to Python 3
The str() in python 2.7 behaves differently than in Python3. In Python 2.7 there is also unicode() which is missing in Python3. So there has to be some thought-through approach to path the source-line in
gnuradio\lib\site-packages\gnuradio\grc\core\params\param.py , line 99
`label = str(labels[i])`

rear1019, 2019-12-30:
> I found the offending string in the file
> `gnuradio\share\gnuradio\grc\blocks\digital_symbol_sync_xx.block.yml`
> 
> The underlying issue is a little deeper. Is the YML file supposed be ASCII or UTF-8?
> Is the data read byte-wise or as strings? Is the encoding handled consistently?
> Further problems ahead: Python 2.7 to Python 3
> The str() in python 2.7 behaves differently than in Python3. In Python 2.7 there is also unicode() which is missing in Python3. So there has to be some thought-through approach to path the source-line in
> gnuradio\lib\site-packages\gnuradio\grc\core\params\param.py , line 99
> `label = str(labels[i])`
> 
> I propose the following patch
> In line 22 it reads:
> 
> > option_labels: ["Mueller and M\xFCller", "Modified Mueller and M\xFCller",
> > If I change it to this, the block can be inserted without eror:
> > option_labels: ["Mueller and Mueller", "Modified Mueller and Mueller", ... ...

With all due respect, the proposed fix is the wrong way to go. The underlying issue(s) should be fixed (see the remarks above), not the symptoms.

Note: The bug does not appear in GNU Radio 3.8 built with Python 3.

RedStone002, 2019-12-30:
I agree.
I'm just not versed enough in GNURadio development to figure out the whole chain of encoding from the yml file to QT.
I don't have access to a py3 build of GNURadio (Windows), but I believe you that the problem will go away. So maybe my patch should not be merged.

mbr0wn, 2020-01-01:
Closed by https://github.com/gnuradio/gnuradio/pull/2986.

closed 2020-01-01 by mbr0wn