gnuradio/gnuradio #2995: Can't launch GNU Radio Companion on Manjaro Linux

PawelPleskaczynski, 2020-01-02:
Hello, I have a small problem with GNU Radio Companion. I built everything from source using this command:
```
cmake -DPYTHON_EXECUTABLE=/usr/bin/python2.7 -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib/libpython2.7.so -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_GRC=ON -DENABLE_GR_QTGUI=ON -DENABLE_INTERNAL_VOLK=OFF DENABLE_GR_CTRLPORT=OFF ../
```
after cloning this repository's `maint-3.7` branch. I want version 3.7 because I need gr-noaa blocks. Everything was built correctly, but after running `make test` I got this output:
```
98% tests passed, 6 tests failed out of 349

Total Test time (real) = 145.43 sec

The following tests FAILED:
        153 - qa_cpp_py_binding (Failed)
        154 - qa_cpp_py_binding_set (Failed)
        155 - qa_ctrlport_probes (Failed)
        194 - qa_skiphead (Failed)
        202 - qa_tag_share (Failed)
        331 - qa_qtgui (Failed)
Errors while running CTest
make: *** [Makefile:152: test] Error 8
```

And after running the `gnuradio-companion` command in the console I got this error message:
[image]

I'm using Manjaro Linux with both Python 3.8 and Python 2.7 installed, and with every needed dependency as well. Should I attach more things to help troubleshoot the problem?

velichkov, 2020-01-03:
Hi @PawelPleskaczynski,

> ```
> 98% tests passed, 6 tests failed out of 349
>
> Total Test time (real) = 145.43 sec
>
> The following tests FAILED:
>         153 - qa_cpp_py_binding (Failed)
>         154 - qa_cpp_py_binding_set [...]

Run `ctest --output-on-failure` from the build directory and provide the output.

> And after running the `gnuradio-companion` command in the console I got this error message:

You are installing in `/usr/` prefix (the `-DCMAKE_INSTALL_PREFIX=/usr` argument) and not in `/usr/local/` so try adding `/usr/lib/python2.7/dist-packages` to the `PYTHONPATH` instead.

> I'm using Manjaro Linux with both Python 3.8 and Python 2.7 installed, and with every needed dependency as well.

The `dist-packages` is a Debian/Ubuntu specific thing and Manjaro is based on Arch so most probably gnuradio is installed in `site-packages` on your system. You could find the location with

```
$ python2 -c "from distutils import sysconfig; print sysconfig.get_python_lib(plat_specific=True, prefix='')"
$ find /usr -name FlowGraph.py
```

PawelPleskaczynski, 2020-01-03:
Hi @velichkov, thank you for responding!
I tried to launch `gnuradio-companion` with `/usr/lib/python2.7/dist-packages` in `PYTHONPATH`, but I got the same error message. I ran the `python2` command you provided and got this output:
```
lib/python2.7/site-packages
```
and then I ran the `find` command:
```
/usr/lib/python3.8/site-packages/gnuradio/grc/core/FlowGraph.py
/usr/lib/python3.8/site-packages/gnuradio/grc/gui/FlowGraph.py
```
When I navigated to `/lib/python2.7/site-packages` I couldn't find gnuradio package, but I could find it in `/usr/lib/python2.7/dist-packages`, I have no idea what's going on.

Here's output of `ctest` command:
```
Test project /home/pawel/gnuradio/build
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
  1/349 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .....................   Passed    0.02 sec
        Start   2: qa_volk_16i_branch_4_state_8
  2/349 Test   #2: qa_volk_16i_branch_4_state_8 .......................   Passed    0.00 sec
        Start   3: qa_volk_16i_convert_8i
  3/349 Test   #3: qa_volk_16i_convert_8i .............................   Passed    0.01 sec
        Start   4: qa_volk_16i_max_star_16i
  4/349 Test   #4: qa_volk_16i_max_star_16i ...........................   Passed    0.00 sec
        Start   5: qa_volk_16i_max_star_horizontal_16i
  5/349 Test   #5: qa_volk_16i_max_star_horizontal_16i ................   Passed    0.01 sec
        Start   6: qa_volk_16i_permute_and_scalar_add
  6/349 Test   #6: qa_volk_16i_permute_and_scalar_add .................   Passed    0.01 sec
        Start   7: qa_volk_16i_s32f_convert_32f
  7/349 Test   #7: qa_volk_16i_s32f_convert_32f .......................   Passed    0.01 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
  8/349 Test   #8: qa_volk_16i_x4_quad_max_star_16i ...................   Passed    0.01 sec
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
  9/349 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .....................   Passed    0.01 sec
        Start  10: qa_volk_16ic_convert_32fc
 10/349 Test  #10: qa_volk_16ic_convert_32fc ..........................   Passed    0.02 sec
        Start  11: qa_volk_16ic_deinterleave_16i_x2
 11/349 Test  #11: qa_volk_16ic_deinterleave_16i_x2 ...................   Passed    0.01 sec
        Start  12: qa_volk_16ic_deinterleave_real_16i
 12/349 Test  #12: qa_volk_16ic_deinterleave_real_16i .................   Passed    0.01 sec
        Start  13: qa_volk_16ic_deinterleave_real_8i
 13/349 Test  #13: qa_volk_16ic_deinterleave_real_8i ..................   Passed    0.01 sec
        Start  14: qa_volk_16ic_magnitude_16i
 14/349 Test  #14: qa_volk_16ic_magnitude_16i .........................   Passed    0.01 sec
        Start  15: qa_volk_16ic_s32f_deinterleave_32f_x2
 15/349 Test  #15: qa_volk_16ic_s32f_deinterleave_32f_x2 ..............   Passed    0.02 sec
        Start  16: qa_volk_16ic_s32f_deinterleave_real_32f
 16/349 Test  #16: qa_volk_16ic_s32f_deinterleave_real_32f ............   Passed    0.01 sec
        Start  17: qa_volk_16ic_s32f_magnitude_32f
 17/349 Test  #17: qa_volk_16ic_s32f_magnitude_32f ....................   Passed    0.01 sec
        Start  18: qa_volk_16ic_x2_dot_prod_16ic
 18/349 Test  #18: qa_volk_16ic_x2_dot_prod_16ic ......................   Passed    0.02 sec
        Start  19: qa_volk_16ic_x2_multiply_16ic
 19/349 Test  #19: qa_volk_16ic_x2_multiply_16ic ......................   Passed    0.02 sec
        Start  20: qa_volk_16u_byteswap
 20/349 Test  #20: qa_volk_16u_byteswap ...............................   Passed    0.01 sec
        Start  21: qa_volk_16u_byteswappuppet_16u
 21/349 Test  #21: qa_volk_16u_byteswappuppet_16u .....................   Passed    0.01 sec
        Start  22: qa_volk_32f_64f_add_64f
 22/349 Test  #22: qa_volk_32f_64f_add_64f ............................   Passed    0.02 sec
        Start  23: qa_volk_32f_64f_multiply_64f
 23/349 Test  #23: qa_volk_32f_64f_multiply_64f .......................   Passed    0.01 sec
        Start  24: qa_volk_32f_8u_polarbutterfly_32f
 24/349 Test  #24: qa_volk_32f_8u_polarbutterfly_32f ..................   Passed    0.01 sec
        Start  25: qa_volk_32f_8u_polarbutterflypuppet_32f
 25/349 Test  #25: qa_volk_32f_8u_polarbutterflypuppet_32f ............   Passed    0.02 sec
        Start  26: qa_volk_32f_accumulator_s32f
 26/349 Test  #26: qa_volk_32f_accumulator_s32f .......................   Passed    0.01 sec
        Start  27: qa_volk_32f_acos_32f
 27/349 Test  #27: qa_volk_32f_acos_32f ...............................   Passed    0.01 sec
        Start  28: qa_volk_32f_asin_32f
[... 652 lines cut ...]
        Start 344: qa_classify
344/349 Test #344: qa_classify ........................................   Passed    0.31 sec
        Start 345: qa_zeromq_pub
345/349 Test #345: qa_zeromq_pub ......................................   Passed    0.28 sec
        Start 346: qa_zeromq_pubsub
346/349 Test #346: qa_zeromq_pubsub ...................................   Passed    0.79 sec
        Start 347: qa_zeromq_pushpull
347/349 Test #347: qa_zeromq_pushpull .................................   Passed    0.77 sec
        Start 348: qa_zeromq_reqrep
348/349 Test #348: qa_zeromq_reqrep ...................................   Passed    0.80 sec
        Start 349: qa_zeromq_sub
349/349 Test #349: qa_zeromq_sub ......................................   Passed    1.02 sec

98% tests passed, 6 tests failed out of 349

Total Test time (real) = 146.96 sec

The following tests FAILED:
	153 - qa_cpp_py_binding (Failed)
	154 - qa_cpp_py_binding_set (Failed)
	155 - qa_ctrlport_probes (Failed)
	194 - qa_skiphead (Failed)
	202 - qa_tag_share (Failed)
	331 - qa_qtgui (Failed)
```

velichkov, 2020-01-03:
> I tried to launch `gnuradio-companion` with `/usr/lib/python2.7/dist-packages` in `PYTHONPATH`, but I got the same error message. I ran the `python2` command you provided and got this output:
>
> ``` [...]

Despite the fact that you've configured it with python2.7 it has been built and installed for python3.8. The cmake has a cache that sometimes could cause problems. First uninstall this, then remove the whole build directory, create it again and reinstall from scratch

```
cd build
sudo make uninstall
cd ..
rm -rvf build
mkdir build
cd build
cmake -DPYTHON_EXECUTABLE=/usr/bin/python2.7 -DPYTHON_INCLUDE_DIR=/usr/include/python2.7 -DPYTHON_LIBRARY=/usr/lib/libpython2.7.so -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_GRC=ON -DENABLE_GR_QTGUI=ON -DENABLE_INTERNAL_VOLK=OFF DENABLE_GR_CTRLPORT=OFF ../
```
Before building make sure the `PYTHON_EXECUTABLE` is set correctly in the `CMakeCache.txt`
```
grep PYTHON_EXECUTABLE CMakeCache.txt
make
sudo make install
```

> When I navigated to `/lib/python2.7/site-packages` I couldn't find gnuradio package, but I could find it in `/usr/lib/python2.7/dist-packages`, I have no idea what's going on.

As you are not using Debian it has been installed in `site-packages` and that's OK. It's in `python3.8` subdirectory and not `python2.7` and that's why python2 could not find it. You need to reinstall it in `/usr/lib/python2.7/site-packages`

> 153/349 Test #153: qa_cpp_py_binding ..................................***Failed    0.22 sec
> Traceback (most recent call last):
>   File [...]

> 194/349 Test #194: qa_skiphead ........................................***Failed    0.54 sec
> /home/pawel/gnuradio/build/gr-blocks/python/blocks/qa_skiphead_test.sh: line 9: 13735 Segmentation fault      (core dumped) /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-blocks/python/blocks/qa_skiphead.py

> 202/349 Test #202: qa_tag_share .......................................***Failed    0.49 sec
> /home/pawel/gnuradio/build/gr-blocks/python/blocks/qa_tag_share_test.sh: line 9: 14064 Segmentation [...]

Let's first fix the python 2/3 problem and then we could take care of these tests if they still fail.

PawelPleskaczynski, 2020-01-03:
Okay, I'll try this right now

PawelPleskaczynski, 2020-01-03:
Okay, it's done. Now I can launch `gnuradio-companion` with a blank project, but when I add QT GUI blocks, the program crashes. Here's output of `ctest` again:
```
Test project /home/pawel/gnuradio/build
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
  1/346 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .....................   Passed    0.02 sec
        Start   2: qa_volk_16i_branch_4_state_8
  2/346 Test   #2: qa_volk_16i_branch_4_state_8 .......................   Passed    0.01 sec
        Start   3: qa_volk_16i_convert_8i
  3/346 Test   #3: qa_volk_16i_convert_8i .............................   Passed    0.01 sec
        Start   4: qa_volk_16i_max_star_16i
  4/346 Test   #4: qa_volk_16i_max_star_16i ...........................   Passed    0.00 sec
        Start   5: qa_volk_16i_max_star_horizontal_16i
  5/346 Test   #5: qa_volk_16i_max_star_horizontal_16i ................   Passed    0.01 sec
        Start   6: qa_volk_16i_permute_and_scalar_add
  6/346 Test   #6: qa_volk_16i_permute_and_scalar_add .................   Passed    0.01 sec
        Start   7: qa_volk_16i_s32f_convert_32f
  7/346 Test   #7: qa_volk_16i_s32f_convert_32f .......................   Passed    0.02 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
  8/346 Test   #8: qa_volk_16i_x4_quad_max_star_16i ...................   Passed    0.01 sec
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
  9/346 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .....................   Passed    0.01 sec
        Start  10: qa_volk_16ic_convert_32fc
 10/346 Test  #10: qa_volk_16ic_convert_32fc ..........................   Passed    0.02 sec
        Start  11: qa_volk_16ic_deinterleave_16i_x2
 11/346 Test  #11: qa_volk_16ic_deinterleave_16i_x2 ...................   Passed    0.01 sec
        Start  12: qa_volk_16ic_deinterleave_real_16i
 12/346 Test  #12: qa_volk_16ic_deinterleave_real_16i .................   Passed    0.01 sec
        Start  13: qa_volk_16ic_deinterleave_real_8i
 13/346 Test  #13: qa_volk_16ic_deinterleave_real_8i ..................   Passed    0.01 sec
        Start  14: qa_volk_16ic_magnitude_16i
 14/346 Test  #14: qa_volk_16ic_magnitude_16i .........................   Passed    0.01 sec
        Start  15: qa_volk_16ic_s32f_deinterleave_32f_x2
 15/346 Test  #15: qa_volk_16ic_s32f_deinterleave_32f_x2 ..............   Passed    0.02 sec
        Start  16: qa_volk_16ic_s32f_deinterleave_real_32f
 16/346 Test  #16: qa_volk_16ic_s32f_deinterleave_real_32f ............   Passed    0.01 sec
        Start  17: qa_volk_16ic_s32f_magnitude_32f
 17/346 Test  #17: qa_volk_16ic_s32f_magnitude_32f ....................   Passed    0.01 sec
        Start  18: qa_volk_16ic_x2_dot_prod_16ic
 18/346 Test  #18: qa_volk_16ic_x2_dot_prod_16ic ......................   Passed    0.02 sec
        Start  19: qa_volk_16ic_x2_multiply_16ic
 19/346 Test  #19: qa_volk_16ic_x2_multiply_16ic ......................   Passed    0.02 sec
        Start  20: qa_volk_16u_byteswap
 20/346 Test  #20: qa_volk_16u_byteswap ...............................   Passed    0.01 sec
        Start  21: qa_volk_16u_byteswappuppet_16u
 21/346 Test  #21: qa_volk_16u_byteswappuppet_16u .....................   Passed    0.01 sec
        Start  22: qa_volk_32f_64f_add_64f
 22/346 Test  #22: qa_volk_32f_64f_add_64f ............................   Passed    0.02 sec
        Start  23: qa_volk_32f_64f_multiply_64f
 23/346 Test  #23: qa_volk_32f_64f_multiply_64f .......................   Passed    0.02 sec
        Start  24: qa_volk_32f_8u_polarbutterfly_32f
 24/346 Test  #24: qa_volk_32f_8u_polarbutterfly_32f ..................   Passed    0.01 sec
        Start  25: qa_volk_32f_8u_polarbutterflypuppet_32f
 25/346 Test  #25: qa_volk_32f_8u_polarbutterflypuppet_32f ............   Passed    0.02 sec
        Start  26: qa_volk_32f_accumulator_s32f
 26/346 Test  #26: qa_volk_32f_accumulator_s32f .......................   Passed    0.01 sec
        Start  27: qa_volk_32f_acos_32f
 27/346 Test  #27: qa_volk_32f_acos_32f ...............................   Passed    0.01 sec
        Start  28: qa_volk_32f_asin_32f
 28/346 Test  #28: qa_volk_32f_asin_32f ...............................   Passed    0.01 sec
        Start  29: qa_volk_32f_atan_32f
 29/346 Test  #29: qa_volk_32f_atan_32f ...............................   Passed    0.02 sec
        Start  30: qa_volk_32f_binary_slicer_32i
 30/346 Test  #30: qa_volk_32f_binary_slicer_32i ......................   Passed    0.01 sec
        Start  31: qa_volk_32f_binary_slicer_8i
 31/346 Test  #31: qa_volk_32f_binary_slicer_8i .......................   Passed    0.01 sec
        Start  32: qa_volk_32f_convert_64f
 32/346 Test  #32: qa_volk_32f_convert_64f ............................   Passed    0.02 sec
        Start  33: qa_volk_32f_cos_32f
 33/346 Test  #33: qa_volk_32f_cos_32f ................................   Passed    0.02 sec
        Start  34: qa_volk_32f_expfast_32f
[... 616 lines cut ...]
339/346 Test #339: qa_ulaw_vocoder ....................................   Passed    0.26 sec
        Start 340: qa_fcd
340/346 Test #340: qa_fcd .............................................   Passed    0.19 sec
        Start 341: qa_classify
341/346 Test #341: qa_classify ........................................   Passed    0.30 sec
        Start 342: qa_zeromq_pub
342/346 Test #342: qa_zeromq_pub ......................................   Passed    0.28 sec
        Start 343: qa_zeromq_pubsub
[... cut ...]
344/346 Test #344: qa_zeromq_pushpull .................................   Passed    0.79 sec
        Start 345: qa_zeromq_reqrep
345/346 Test #345: qa_zeromq_reqrep ...................................   Passed    0.78 sec
        Start 346: qa_zeromq_sub
346/346 Test #346: qa_zeromq_sub ......................................   Passed    1.00 sec

99% tests passed, 3 tests failed out of 346

Total Test time (real) = 147.33 sec

The following tests FAILED:
	191 - qa_skiphead (Failed)
	199 - qa_tag_share (Failed)
	328 - qa_qtgui (Failed)
```

velichkov, 2020-01-03:
> Okay, it's done. Now I can launch `gnuradio-companion` with a blank project,

:+1: 

> but when I add QT GUI blocks, the program crashes. Here's output of `ctest` again:
> 191/346 Test #191: qa_skiphead ........................................***Failed    0.56 sec
> /home/pawel/gnuradio/build/gr-blocks/python/blocks/qa_skiphead_test.sh: line 9:  2382 Segmentation fault      (core dumped) /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-blocks/python/blocks/qa_skiphead.py

You need to take a backtrace with `gdb`. Open `/home/pawel/gnuradio/build/gr-blocks/python/blocks/qa_skiphead_test.sh` and change the last line to

```bash
gdb --ex run --args /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-blocks/python/blocks/qa_skiphead.py
```
Then in the terminal execute the `qa_skiphead_test.sh` and when it crashes you will see the `(gdb)` prompt, execute `bt` command and provide the full output.

If it does not print filenames and line numbers you may need to rebuild with `-DCMAKE_BUILD_TYPE=RelWithDebInfo`

> 199/346 Test #199: qa_tag_share .......................................***Failed    0.49 sec
> /home/pawel/gnuradio/build/gr-blocks/python/blocks/qa_tag_share_test.sh: line 9:  2714 Segmentation [...]

> 328/346 Test #328: qa_qtgui ...........................................***Failed    0.41 sec
> /home/pawel/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh: line 9: 21730 Segmentation fault      (core dumped) /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-qtgui/python/qtgui/qa_qtgui.py

Same procedure as above for `qa_tag_share_test.sh` and `qa_qtgui_test.sh` scripts.

PawelPleskaczynski, 2020-01-03:
Okay, here are the outputs 
`qa_skipahead_test.sh`:
```
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python2.7...
(No debugging symbols found in /usr/bin/python2.7)
Starting program: /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-blocks/python/blocks/qa_skiphead.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
/usr/lib/../share/gcc-9.2.0/python/libstdcxx/v6/xmethods.py:731: SyntaxWarning: list indices must be integers or slices, not str; perhaps you missed a comma?
  refcounts = ['_M_refcount']['_M_pi']
[New Thread 0x7ffff002d700 (LWP 24431)]
[New Thread 0x7fffef82c700 (LWP 24432)]
[New Thread 0x7fffef02b700 (LWP 24433)]
[New Thread 0x7fffee82a700 (LWP 24434)]
[Thread 0x7fffef82c700 (LWP 24432) exited]
[Thread 0x7ffff002d700 (LWP 24431) exited]
[Thread 0x7fffef02b700 (LWP 24433) exited]
[Thread 0x7fffee82a700 (LWP 24434) exited]
[New Thread 0x7fffee82a700 (LWP 24435)]
[New Thread 0x7fffef02b700 (LWP 24436)]
[New Thread 0x7fffef82c700 (LWP 24437)]
[New Thread 0x7ffff002d700 (LWP 24438)]
[Thread 0x7fffee82a700 (LWP 24435) exited]
[Thread 0x7fffef02b700 (LWP 24436) exited]
[Thread 0x7fffef82c700 (LWP 24437) exited]
[Thread 0x7ffff002d700 (LWP 24438) exited]
[New Thread 0x7ffff002d700 (LWP 24439)]
[New Thread 0x7fffef82c700 (LWP 24440)]
[New Thread 0x7fffef02b700 (LWP 24441)]
[New Thread 0x7fffee82a700 (LWP 24442)]
[Thread 0x7fffef02b700 (LWP 24441) exited]
[Thread 0x7fffef82c700 (LWP 24440) exited]
[Thread 0x7ffff002d700 (LWP 24439) exited]
[Thread 0x7fffee82a700 (LWP 24442) exited]
[New Thread 0x7fffee82a700 (LWP 24443)]
[New Thread 0x7fffef02b700 (LWP 24444)]
[New Thread 0x7fffef82c700 (LWP 24445)]
[New Thread 0x7ffff002d700 (LWP 24446)]
[Thread 0x7fffee82a700 (LWP 24443) exited]
[Thread 0x7fffef82c700 (LWP 24445) exited]
[Thread 0x7ffff002d700 (LWP 24446) exited]
[Thread 0x7fffef02b700 (LWP 24444) exited]
[New Thread 0x7ffff002d700 (LWP 24447)]
[New Thread 0x7fffef82c700 (LWP 24448)]
[New Thread 0x7fffef02b700 (LWP 24449)]
[New Thread 0x7fffee82a700 (LWP 24450)]
[Thread 0x7fffef02b700 (LWP 24449) exited]
[Thread 0x7fffef82c700 (LWP 24448) exited]
[Thread 0x7fffee82a700 (LWP 24450) exited]
[Thread 0x7ffff002d700 (LWP 24447) exited]
[New Thread 0x7fffee82a700 (LWP 24451)]
[New Thread 0x7fffef02b700 (LWP 24452)]
[New Thread 0x7fffef82c700 (LWP 24453)]
[New Thread 0x7ffff002d700 (LWP 24454)]
[Thread 0x7fffef82c700 (LWP 24453) exited]
[... 193 lines cut ...]
#28 0x00007ffff7c5a9ea in builtin___import__.lto_priv.0 () at /usr/lib/libpython2.7.so.1.0
#29 0x00007ffff7ca8193 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#30 0x00007ffff7c4d662 in PyEval_CallObjectWithKeywords () at /usr/lib/libpython2.7.so.1.0
#31 0x00007ffff7c52205 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#32 0x00007ffff7c4d9c6 in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#33 0x00007ffff7c4e877 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#34 0x00007ffff7cc6362 in PyImport_ExecCodeModuleEx () at /usr/lib/libpython2.7.so.1.0
#35 0x00007ffff7cc67c7 in load_compiled_module () at /usr/lib/libpython2.7.so.1.0
#36 0x00007ffff7cc6f63 in load_package () at /usr/lib/libpython2.7.so.1.0
#37 0x00007ffff7cc6c9a in import_submodule.lto_priv () at /usr/lib/libpython2.7.so.1.0
#38 0x00007ffff7cc567a in ensure_fromlist.part.0.lto_priv () at /usr/lib/libpython2.7.so.1.0
#39 0x00007ffff7c3f704 in PyImport_ImportModuleLevel () at /usr/lib/libpython2.7.so.1.0
#40 0x00007ffff7c5a9ea in builtin___import__.lto_priv.0 () at /usr/lib/libpython2.7.so.1.0
#41 0x00007ffff7ca8193 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#42 0x00007ffff7c4d662 in PyEval_CallObjectWithKeywords () at /usr/lib/libpython2.7.so.1.0
#43 0x00007ffff7c52205 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#44 0x00007ffff7c4d9c6 in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#45 0x00007ffff7c4e877 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#46 0x00007ffff7cc31c2 in run_mod () at /usr/lib/libpython2.7.so.1.0
#47 0x00007ffff7cc326a in PyRun_FileExFlags () at /usr/lib/libpython2.7.so.1.0
#48 0x00007ffff7cc3c92 in PyRun_SimpleFileExFlags () at /usr/lib/libpython2.7.so.1.0
#49 0x00007ffff7cc0dba in Py_Main () at /usr/lib/libpython2.7.so.1.0
#50 0x00007ffff7de1153 in __libc_start_main () at /usr/lib/libc.so.6
#51 0x000055555555505e in _start ()
```

velichkov, 2020-01-03:
> `qa_skipahead_test.sh`:

> ```
> Thread 1 "python2.7" received signal SIGSEGV, Segmentation fault.
> 0x00007ffff6e3ea96 in pmt::is_bool(boost::intrusive_ptr<pmt::pmt_base>) () from [...]

No file names and line numbers unfortunately. Could you do a clean rebuild with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` and then take again a backtrace.

> `qa_tag_share_test.sh`:

> ```
> Thread 1 "python2.7" received signal SIGSEGV, Segmentation fault.
> 0x00007ffff6e3ea96 in pmt::is_bool(boost::intrusive_ptr<pmt::pmt_base>) () from [...]

It's similar to the above one.

> `qa_qtgui_test.sh`:
> 
> ```
> Program received signal SIGSEGV, Segmentation fault.
> 0x00007fffef9f86e8 in QMetaObject::className() const () from /usr/lib/libQt5Core.so.5
> (gdb) bt
> #0  0x00007fffef9f86e8 in QMetaObject::className() const () at /usr/lib/libQt5Core.so.5
> #1  0x00007ffff04e3cf8 in  () at /usr/lib/libQt5Widgets.so.5
> #2  0x00007ffff047f997 in  () at /usr/lib/libQt5Widgets.so.5
> ```

Could you install the Qt5 debug info package/symobls and then take again a backtrace. I'm not familiar with Arch/Manjaro linux and I don't know exactly how to do it. 

On fedora you could install the debug info with `sudo debuginfo-install qt5-qtbase-devel` and that installs the `qt5-qtbase-devel-debuginfo` package.

PawelPleskaczynski, 2020-01-03:
Oh, I thought this version of GNU Radio uses PyQt4? I have PyQt4 installed for Python 2.7, but I didn't install PyQt5. I'll try to do that first

PawelPleskaczynski, 2020-01-03:
Well, it crashes anyways, so I'll do what you wrote

PawelPleskaczynski, 2020-01-03:
Done, but unfortunately there are no debug packages in Arch/Manjaro repositories
`qa_skiphead_test.sh`:
```
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python2.7...
(No debugging symbols found in /usr/bin/python2.7)
Starting program: /usr/bin/python2.7 -B /home/pawel/gnuradio/gr-blocks/python/blocks/qa_skiphead.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
/root/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available
[New Thread 0x7ffff0387700 (LWP 14476)]
[New Thread 0x7fffefb86700 (LWP 14477)]
[New Thread 0x7fffef385700 (LWP 14478)]
[New Thread 0x7fffeeb44700 (LWP 14479)]
[Thread 0x7ffff0387700 (LWP 14476) exited]
[Thread 0x7fffefb86700 (LWP 14477) exited]
[Thread 0x7fffef385700 (LWP 14478) exited]
[Thread 0x7fffeeb44700 (LWP 14479) exited]
[New Thread 0x7fffeeb44700 (LWP 14480)]
[New Thread 0x7fffef385700 (LWP 14481)]
[New Thread 0x7fffefb86700 (LWP 14482)]
[New Thread 0x7ffff0387700 (LWP 14483)]
[Thread 0x7fffeeb44700 (LWP 14480) exited]
[Thread 0x7fffefb86700 (LWP 14482) exited]
[Thread 0x7fffef385700 (LWP 14481) exited]
[Thread 0x7ffff0387700 (LWP 14483) exited]
[New Thread 0x7ffff0387700 (LWP 14484)]
[New Thread 0x7fffefb86700 (LWP 14485)]
[New Thread 0x7fffef385700 (LWP 14486)]
[New Thread 0x7fffeeb44700 (LWP 14487)]
[Thread 0x7ffff0387700 (LWP 14484) exited]
[Thread 0x7fffef385700 (LWP 14486) exited]
[Thread 0x7fffefb86700 (LWP 14485) exited]
[Thread 0x7fffeeb44700 (LWP 14487) exited]
[New Thread 0x7fffeeb44700 (LWP 14488)]
[New Thread 0x7fffef385700 (LWP 14489)]
[New Thread 0x7fffefb86700 (LWP 14490)]
[New Thread 0x7ffff0387700 (LWP 14491)]
[Thread 0x7fffefb86700 (LWP 14490) exited]
[Thread 0x7fffef385700 (LWP 14489) exited]
[Thread 0x7ffff0387700 (LWP 14491) exited]
[Thread 0x7fffeeb44700 (LWP 14488) exited]
[New Thread 0x7ffff0387700 (LWP 14492)]
[New Thread 0x7fffefb86700 (LWP 14493)]
[New Thread 0x7fffef385700 (LWP 14494)]
[New Thread 0x7fffeeb44700 (LWP 14495)]
[Thread 0x7fffef385700 (LWP 14494) exited]
[Thread 0x7fffefb86700 (LWP 14493) exited]
[Thread 0x7ffff0387700 (LWP 14492) exited]
[Thread 0x7fffeeb44700 (LWP 14495) exited]
[New Thread 0x7fffeeb44700 (LWP 14496)]
[New Thread 0x7fffef385700 (LWP 14497)]
[New Thread 0x7fffefb86700 (LWP 14498)]
[New Thread 0x7ffff0387700 (LWP 14499)]
[Thread 0x7fffefb86700 (LWP 14498) exited]
[... 337 lines cut ...]
#28 0x00007ffff7c5a9ea in builtin___import__.lto_priv.0 () at /usr/lib/libpython2.7.so.1.0
#29 0x00007ffff7ca8193 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#30 0x00007ffff7c4d662 in PyEval_CallObjectWithKeywords () at /usr/lib/libpython2.7.so.1.0
#31 0x00007ffff7c52205 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#32 0x00007ffff7c4d9c6 in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#33 0x00007ffff7c4e877 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#34 0x00007ffff7cc6362 in PyImport_ExecCodeModuleEx () at /usr/lib/libpython2.7.so.1.0
#35 0x00007ffff7cc67c7 in load_compiled_module () at /usr/lib/libpython2.7.so.1.0
#36 0x00007ffff7cc6f63 in load_package () at /usr/lib/libpython2.7.so.1.0
#37 0x00007ffff7cc6c9a in import_submodule.lto_priv () at /usr/lib/libpython2.7.so.1.0
#38 0x00007ffff7cc567a in ensure_fromlist.part.0.lto_priv () at /usr/lib/libpython2.7.so.1.0
#39 0x00007ffff7c3f704 in PyImport_ImportModuleLevel () at /usr/lib/libpython2.7.so.1.0
#40 0x00007ffff7c5a9ea in builtin___import__.lto_priv.0 () at /usr/lib/libpython2.7.so.1.0
#41 0x00007ffff7ca8193 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#42 0x00007ffff7c4d662 in PyEval_CallObjectWithKeywords () at /usr/lib/libpython2.7.so.1.0
#43 0x00007ffff7c52205 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#44 0x00007ffff7c4d9c6 in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#45 0x00007ffff7c4e877 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#46 0x00007ffff7cc31c2 in run_mod () at /usr/lib/libpython2.7.so.1.0
#47 0x00007ffff7cc326a in PyRun_FileExFlags () at /usr/lib/libpython2.7.so.1.0
#48 0x00007ffff7cc3c92 in PyRun_SimpleFileExFlags () at /usr/lib/libpython2.7.so.1.0
#49 0x00007ffff7cc0dba in Py_Main () at /usr/lib/libpython2.7.so.1.0
#50 0x00007ffff7de1153 in __libc_start_main () at /usr/lib/libc.so.6
#51 0x000055555555505e in _start ()
```

velichkov, 2020-01-03:
Would it be possible to reformat your posts and use [`<details><summary>`](https://stackoverflow.com/questions/52214187/how-can-i-fold-content-in-github-markdown/52215506#52215506) blocks as these long backtrackes make the discussion hard to follow
```
<details>
  <summary>Click to expand</summary>
  whatever
</details>
```
> unfortunately there are no debug packages in Arch/Manjaro repositories

See:
* https://wiki.archlinux.org/index.php/Debug_-_Getting_Traces
* https://wiki.archlinux.org/index.php/Unofficial_user_repositories#qt-debug

PawelPleskaczynski, 2020-01-03:
Okay, I reformatted my posts, sorry for the trouble. I'll read articles you linked and I'll try to install debug packages (weirdly, when I was looking for Qt debug package with my AUR helper, it didn't come up, so I thought it just wasn't present)

velichkov, 2020-01-06:
Hi @PawelPleskaczynski,

> I reformatted my posts, sorry for the trouble.

Thanks.

I was able to reproduce the failure in `qa_skiphead` and `qa_tag_share` tests, and applying d6e217cbc00ec0333ba030ab2d84c31b7dbb77d5 on `maint-3.7` branch resolve the problem.
```
git cherry-pick d6e217cbc00ec0333ba030ab2d84c31b7dbb77d5
``` 

> I thought this version of GNU Radio uses PyQt4? 

I don't know. I'm not able to reproduce the failure in `qa_qtgui_test` :(, on my system this test seems to not load this `libQt5Widgets.so.5` library.

PawelPleskaczynski, 2020-01-06:
Hi @velichkov,
I'll try to apply this commit tomorrow and I'll let you know if it helped to solve these two failed tests. I'm not sure about the Qt failure though, maybe there's something wrong with my Linux installation? I'll try it on a different system in a VM.
Anyways, thank you very much for your help!

PawelPleskaczynski, 2020-01-10:
Hi @velichkov,
I managed to successfully build GNU Radio with GRC from source in Kubuntu. Everything works well, including Qt GUI, and 99% of the tests were passed. `qa_zeromq_pub`, `qa_zeromq_pubsub`, `qa_zeromq_pushpull`, `qa_zeromq_reqrep` and `qa_zeromq_sub` failed, but it doesn't seem to mess with program's behavior, at least for my usage. It seems that I had to overlook installation of some dependencies on my main Manjaro Linux installation. Thank you very much for help, I think this issue can be closed

closed 2020-01-10 by PawelPleskaczynski