gnuradio/gnuradio #3004: Replace Swig with PyBind11

devnulling, 2020-01-04:
Place holder for 3.9 Project

mormj, 2020-01-10:
Currently in progress - a prototype implementing the new pybind11 bindings on a few blocks 
github.com/mormj/gnuradio  pybind11 branch

Challenges:
1. Pybind11 does not automatically convert header files as in swig, so all of the binding must be done as c++ macros (for this the header parsing blocktool may come in handy)
2. Not sure pybind works great with boost smart pointers, but removal of these has already been put in motion - right now have cherry picked the commits that switch to std::shared_ptr
3.  There are a **lot** of blocks in-tree, so doing all this manually and validating is a lot of work and needs a methodical approach

Development items (add to this list as we understand the problem more)
- [x] Removal of all swig
- [ ] Agreement on the code and structure for binding code - right now have modeled after UHD
- [ ] Clean up / remove the gateway - can pybind11 do more of this than swig could.  Get a basic python flowgraph running
- [ ] Tool for generating pybind headers - base it on blocktool?
- [ ] CMake for grabbing copied in-tree pybind headers - right now I'm just installing pybind11 to the prefix and referencing it that way
- [ ] Modtool generate new structure

Current Issues (will track on mormj:pybind11)
- [ ] Had to put the to_basic_block() up at the derived class for pybind11 to properly upcast

skoslowski, 2020-01-10:
* pybind does have support and helpers for custom smart ptrs. However, I'd prefer the switch to the ptr types to std.
* I don't like the terms "exports" and "export" here. How about something like wrappers or bindings?
* I suggest to make the wrapped return type of `to_basic_block` binding framework agnostic, allowing OOTs to use something other than pybind11. Sth. based on pycapsule or a custom type defined as part of the runtime with a c-api (that custom type may in turn be implemented using pybind)

mbr0wn, 2020-01-10:
Yeah we all want std:: pointers :+1:

mormj, 2020-01-23:
For the general structure, I'm currently running with the following (similar to pyuhd, and renamed as suggested):
[image]
python/[module]/
- bindings (directory to hold all pybind related code)
    - custom 
        - headername_python.hpp (e.g. multiply_const_python.hpp)
        - ...  all public headers with bound methods that had to be done manually
    - generated
        - ... automatically generated headers (e.g. null_source_python.hpp)
    - python_bindings.cpp 
        - (includes each of either custom or generated header file, and calls the appropriate bind_ function)

I was able to get around the to_basic_block nonsense by making a lambda function in the pybind definition for the block (not in the top level header file), so that is much cleaner.  So now if an OOT wants to bind a different way, nothing has changed in the API, and this change is limited to the pybind specific code.

Current issue is that pygccxml (which blocktool uses under the hood) does not populate template classes with enough info to generate the expanded classes (e.g. multiply_const_{ss,ii,ff,cc}).   So these have to be done manually.  There are about 30 of them in gr-blocks so it's not terrible, but for OOTs it will be a manual step that we should be able to identify when they run modtool/blocktool and flag that the bindings will have to be done manually.

mormj, 2020-01-23:
Tracking progress more dynamically here:
https://github.com/mormj/gnuradio/projects/1

mbr0wn, 2020-09-09:
Pybind11 is in master.

closed 2020-09-09 by mbr0wn