gnuradio/gnuradio #3033: qtgui: waterfall plots that don't do the fft internally

mormj, 2020-01-06:
The QT gui waterfall sink blocks perform and fft and a magnitude calculation internally.  Sometimes FFT is already done in hardware but we want to throw on a waterfall plot to see the result.  

Add an option in the waterfall plots to bypass performing the fft/magnitude, and take vectors of fft or fft/magnitude

perhaps this a new block altogether so as to not complicate the current waterfall block

dkozel, 2020-01-08:
I think a new block with most of the internals shared makes sense. That would also keep it easier in the future for graphical improvements to the waterfall, currently the default visualization is really poor.
QT GUI Vector Waterfall Sink

devnulling, 2020-01-19:
So the Time Raster plot sort of does this functionality already. 

Perhaps it'd be better to try to combine them or reuse some of the code of the Time Raster sink for this, or replace it, as there seems to be some other performance delta between the QT Waterfall sink  and the Time Raster sink. related issue: https://github.com/gnuradio/gnuradio/issues/1120

Also it'd be great to fix the QT waterfall so it "falls" instead of "rises".