gnuradio/gnuradio #3035: make test qa_zeromq_sub fails

Milind-Blaze, 2020-01-07:
Hi! I'm building gnuradio from source as described [here](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Building_and_installing_UHD_from_source_code). Upon running 

```make test```

I received the following results.

99% tests passed, 1 tests failed out of 346

Total Test time (real) = 126.87 sec

The following tests FAILED:
	343 - qa_zeromq_sub (Failed)
Errors while running CTest
Makefile:61: recipe for target 'test' failed
make: *** [test] Error 8

Running ```ctest -V -R qa_zeromq_sub``` results in

UpdateCTestConfiguration  from :/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/build/DartConfiguration.tcl
Test project /home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 343
    Start 343: qa_zeromq_sub

343: Test command: /bin/sh "/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/build/gr-zeromq/python/zeromq/qa_zeromq_sub_test.sh"
343: Test timeout computed to be: 9.99988e+06
343: FF
343: ======================================================================
343: FAIL: test_001 (__main__.qa_zeromq_sub)
343: ----------------------------------------------------------------------
343: Traceback (most recent call last):
343:   File "/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/gr-zeromq/python/zeromq/qa_zeromq_sub.py", line 58, in test_001
343:     self.assertFloatTuplesAlmostEqual(sink.data(), src_data)
343:   File "/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 88, in assertFloatTuplesAlmostEqual
343:     self.assertEqual (len(a), len(b))
343: AssertionError: 0 != 1000
343: 
343: ======================================================================
343: FAIL: test_002 (__main__.qa_zeromq_sub)
343: ----------------------------------------------------------------------
343: Traceback (most recent call last):
343:   File "/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/gr-zeromq/python/zeromq/qa_zeromq_sub.py", line 79, in test_002
343:     self.assertFloatTuplesAlmostEqual(sink.data(), expected_data)
343:   File "/home/milind/projects/5Gtestbed/USRP/toolchain/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 88, in assertFloatTuplesAlmostEqual
343:     self.assertEqual (len(a), len(b))
343: AssertionError: 0 != 2000
343: 
343: ----------------------------------------------------------------------
343: Ran 2 tests in 0.612s
343: 
343: FAILED (failures=2)
1/1 Test #343: qa_zeromq_sub ....................***Failed    0.79 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.80 sec

The following tests FAILED:
	343 - qa_zeromq_sub (Failed)
Errors while running CTest

This is gnuradio v3.7.13.4. How does one fix this?

alexcolpitts96, 2020-01-20:
What operating system are you using?

michaelld, 2020-01-20:
Whether ZMQ works or not (in general as well as in GR specifically) very much depends on the OS & its version, and the version of ZMQ ... and, how your OS's networking & firewall are configured. There are quite a few variables to take into account! ZMQ in GR has bugs, and we're trying to work through them to make gr-zeromq more reliable. Hence, whatever info about your OS & general setup you can provide would be useful.

nareo2, 2020-01-22:
Hi while I'm follwing guide in [here](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux), I got an same issue only qa_zeromq_sub failed

My OS is ubuntu 16.04, zmq.__version__ is 17.1.2 and zmq.zmq_version() is 4.2.5 

Thanks.

nareo2, 2020-01-22:
Hi while I'm following an same [link](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux), I also failed only qa_zeromq_sub and the result of `ctest -V -R qa_zeromq_sub` is same

My dependency is here
Os  : ubuntu 16.04
libzmq : 4.1.4
pyzmq : 15.2.0
GNU Radio : v3.7.13.4

If you need any dependencies please let me know, thank!

alexcolpitts96, 2020-01-22:
I will try it on my Ubuntu 16.04 VM later but try the dependencies [here](https://wiki.gnuradio.org/index.php/UbuntuInstall#Installing_GNU_Radio). 

I have always had success from that page.

nareo2, 2020-01-23:
It worked! 
My [link](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux) use `apt-get install python-zmq` and your not.
So I tried to install pyzmq with `pip install pyzmq` and the test passed
Thank you alexcolpitts96
FYI passed dependencies is here
libzmq: 4.3.2
pyzmq : 18.1.1

alexcolpitts96, 2020-01-23:
@nareo2 I am glad it worked!

@Milind-Blaze, have you had any success?

Milind-Blaze, 2020-01-26:
Hi! No, I tried a few things but couldn't get it to work. Therefore, I installed gnuradio despite the error. 

Here are the necessary details
Ubuntu 16.04
GNU Radio v3.7.13.4
libzmq 4.2.5
pyzmq 17.1.2

Please let me know if any other details are required and I shall update them duly.

alexcolpitts96, 2020-01-27:
Did you try using the maint-3.7 branch as suggested by the Ettus source you used?

Did you GNU Radio installation work despite the failed test? It could be a test issue if you have all the dependencies.

Yacinetel, 2020-05-18:
it's work for me too

marcusmueller, 2024-01-08:
cannot be reproduced anymore

closed 2024-01-08 by marcusmueller