gnuradio/gnuradio #3054: SWIG uses system headers intead of tree headers

mbr0wn, 2020-01-09:
SWIG doesn't use the header files in the source tree, rather, it uses headers it can find in the system. This means every time the system headers and the tree headers differ, SWIG will fail with a mysterious error.

ekigwana, 2020-01-09:
From _gr-digital/swig_ sample output of the generator expression for [GR_SWIG_INCLUDE_DIRS](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/swig/CMakeLists.txt#L26) below, it looks like we have the following items to consider:

- [ ] Missing tree module header directory e.g /home/ekigwana/ext/gnuradio/gr-digital/include/
- [ ] System headers before volk tree headers
- [ ] Duplicates

```
/home/ekigwana/ext/gnuradio/build/gnuradio-runtime/swig;
/home/ekigwana/ext/gnuradio/gnuradio-runtime/swig;
/home/ekigwana/ext/gnuradio/build/gnuradio-runtime/swig;
/home/ekigwana/ext/gnuradio/gnuradio-runtime/swig;
/home/ekigwana/ext/gnuradio/gnuradio-runtime/swig;
/home/ekigwana/ext/gnuradio/build/gnuradio-runtime/swig;
;
/usr/include/python3.7m;
;
/home/ekigwana/ext/gnuradio/gnuradio-runtime/lib/../include;
/home/ekigwana/ext/gnuradio/build/gnuradio-runtime/lib/../include;
/home/ekigwana/ext/gnuradio/gnuradio-runtime/lib/pmt/../../include;
/usr/include;
/usr/include;
/usr/include;
/usr/include;
/usr/include;
/usr/include;
/home/ekigwana/ext/gnuradio/build/volk/include;
/home/ekigwana/ext/gnuradio/volk/include;
/home/ekigwana/ext/gnuradio/volk/kernels;
/usr/include;
/usr/include;
/usr/include;
/usr/include;
/usr/include
```

michaelld, 2020-01-10:
Interesting list! How did you get CMake to print it out, since it's a generator expression?

ekigwana, 2020-01-10:
Added a custom Target to the gr-digital/swig CMakeLists.txt to echo the var. Then build the target to get it to print. You can also write to a file https://stackoverflow.com/questions/51353110/how-do-i-output-the-result-of-a-generator-expression-in-cmake

ekigwana, 2020-01-10:
Since Swig is being replaced with PyBind11 per #3004 It is not unreasonable to uninstall prior to rebuilding. However, PyBind11 also has the same issue and perhaps I can help with that effort instead.

michaelld, 2020-01-10:
SWIG -> PyBind11 should be for just GR39 I think, since it's a huge change to GR API & how OOTs work. So ... we we'll still want to fixup GR38 / SWIG if it's possible to do so. And backport to GR37 if reasonably possible too.

michaelld, 2020-01-10:
The basic issue is how CMake orders `include_directories`; IIRC there are 4 ways to specify them: via `include_directories` which is global from this directory onward (but not parent directories IIRC), and easy to control; the local directory only via a property, which is also easy to control; the in-tree TARGET via a property, which is also easy to control; dependency TARGET via a property, which isn't simple to control. The ordering here is critical, and my memory from testing in early December is that the default CMake ordering puts the final property (dependency TARGET) -before- at least one of the others, which is a bug IMHO. That said, my memory is now ~1 month old, and I have other issues of higher priority at the moment ... so I might be off a little in what I write. I'm adding this note for completion's sake & hoping its useful!

karel, 2020-02-25:
I just got an error `error: 'class gr::digital::pfb_clock_sync_ccf' has no member named 'set_taps'` when compiling GNU Radio from master. Apparently the error is related to this issue here. Removing the installed version of GNU Radio helped.

michaelld, 2020-02-26:
@karel as you found: removing any already-installed GNU Radio takes care of the issue. The issue within CMake / GR isn't easy to track down nor fix, at least in my experience.

schneider42, 2020-09-18:
Uninstalling an OOT which is in such a state can be tricky if one didn't know about the bug, cleaned the build directory and lost `CMakeFiles/uninstall`. It took me about 1h from hitting this bug until arriving here :/

schneider42, 2020-09-18:
My quick fix for `gr-iridium` looks like this:
```patch
diff --git a/swig/CMakeLists.txt b/swig/CMakeLists.txt
index 01c5808131..5b71a4fef7 100644
--- a/swig/CMakeLists.txt
+++ b/swig/CMakeLists.txt
@@ -40,7 +40,7 @@ include(GrPython)
 ########################################################################
 # Setup swig generation
 ########################################################################
-set(GR_SWIG_INCLUDE_DIRS $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>)
+set(GR_SWIG_INCLUDE_DIRS "../include" $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>)
 set(GR_SWIG_TARGET_DEPS gnuradio::runtime_swig)
 
 set(GR_SWIG_LIBRARIES gnuradio-iridium)
```

marcusmueller, 2024-01-08:
closing as maint-3.8 / swig era

closed 2024-01-08 by marcusmueller