gnuradio/gnuradio #3071: doxygen build MSCGEN_PATH and PERL_PATH are obsolete now

ValZapod, 2020-01-13:
So, there is a problem about those https://github.com/gnuradio/gnuradio/search?q=PERL_PATH&unscoped_q=PERL_PATH and those should not be there. You can use doxygen -u to update those file or just delete those variables.

ValZapod, 2020-01-13:
@michaelld Apply this diff on top of master. 
[doxy.txt](https://github.com/gnuradio/gnuradio/files/4055498/doxy.txt)
Produced by doxygen -u <all files mention above in github search> (
It indeed fixed all warnings about MSCGEN_PATH and PERL_PATH.

ValZapod, 2020-01-13:
https://github.com/gnuradio/gnuradio/blob/96a5e82123ddc8a3ea6ced7bb903cafacd5113fb/gr-channels/include/gnuradio/channels/selective_fading_model2.h#L59

This should be delays_std and then delays_maxdev. 
Also \param mags is used twice which broke it.

ValZapod, 2020-01-13:
```
./gnuradio/gnuradio-runtime/include/gnuradio/thrift_application_base.h:158: warning: unable to resolve reference to 'page_logger' for \ref command
./gnuradio/gr-uhd/include/gnuradio/uhd/usrp_sink.h:58: warning: unable to resolve reference to 'page_tagged_stream_blocks' for \ref command
./gnuradio/gr-digital/docs/packet_comms.dox:164: warning: unable to resolve reference to 'page_tagged_stream_blocks' for \ref command
./gnuradio/gr-digital/docs/packet_comms.dox:622: warning: unable to resolve reference to 'page_pmt' for \ref command
./gnuradio/gr-fec/docs/fec.dox:131: warning: unable to resolve reference to 'page_tagged_stream_blocks' for \ref command
./gnuradio/gr-fec/docs/fec.dox:173: warning: unable to resolve reference to 'page_msg_passing' for \ref command
./gnuradio/gr-filter/docs/filter.dox:59: warning: unable to resolve reference to 'pfb_arb_resampler' for \ref command
./gnuradio/gr-uhd/docs/uhd.dox:34: warning: unable to resolve reference to 'msg_passing_commands' for \ref command

```

michaelld, 2020-01-14:
@ValZapod do you know what doxygen version these changes occurred in? If we update & fix the doxygen issues, which I think would be great to do, then we also need to make sure the doxygen found is new enough. I really don't know how far back one has to go for the changes to not work (the minimum required doxygen version) ...

ValZapod, 2020-01-14:
@michaelld https://github.com/doxygen/doxygen/commit/6d1535c38fe6bdaa2a00fff0e7e43774a740a4ce (as you can see the first file is default Doxyfile from that values are taken)
https://github.com/doxygen/doxygen/commit/873e0ccfbe43ef98d4858beb08eef37631ae3cf6
Both in version 1.8.17, perl really not in use since 1.8.0.
Anyway, it is safe as previous versions of doxygen will just use default values for it.

ValZapod, 2020-01-15:
@michaelld See my branch in my fork doxygen-fixes issues. Any comments? In particular is think here it looks better https://www.gnuradio.org/doc/doxygen/selective__fading__model_8h_source.html but i suppoose we need to use only 80 symbols? Watched twitch, BTW, would be nice if you will use 1080p60 ;)

michaelld, 2020-01-15:
> @michaelld [doxygen/doxygen@6d1535c](https://github.com/doxygen/doxygen/commit/6d1535c38fe6bdaa2a00fff0e7e43774a740a4ce) (as you can see the first file is default Doxyfile from that values are taken)
> [doxygen/doxygen@873e0cc](https://github.com/doxygen/doxygen/commit/873e0ccfbe43ef98d4858beb08eef37631ae3cf6)
> Both in version 1.8.17, perl really not in use since 1.8.0.
> Anyway, it is safe as previous versions of doxygen will just use default values for it.

Ah; OK this all makes sense! Let me look at the diff ...

michaelld, 2020-01-15:
I think the Twitch thing was a trial to see if/how well it might work. It would be cool to do more of them, and assuming "we" do then we'll try to tweak things such as the resolution. I'd also love to see multi-person video-in-video: doing PRs and chatting. Martin is an awesome guy & always fun to chat with; be great to get Marcus in there as well & some other of the GR developers.

ValZapod, 2020-01-15:
@michaelld So all of those   

> warning: Included by graph for 

were because of this big file https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/include/gnuradio/block.h included in all other very small files, we need to increase all  DOT_GRAPH_MAX_NODES in all files maybe in 2-3 times.

Also, a question, what doxygen docs is this https://www.gnuradio.org/doc/doxygen/index.html based on (version of GR code)? 

`for \ref command
`
are because of html documentation (for example this https://www.gnuradio.org/doc/doxygen/page_tagged_stream_blocks.html, it should be `\subpage page_tagged_stream_blocks` not not `\ref page_tagged_stream_blocks`) some others (https://www.gnuradio.org/doc/doxygen/page_pmt.html) do not exist anymore (?). I will open pull request and maybe we will discuss it further.

michaelld, 2020-01-15:
> Also, a question, what doxygen docs is this https://www.gnuradio.org/doc/doxygen/index.html based on (version of GR code)?

I honestly don't know. Let me see if I can track down that change in our website repo ...

ValZapod, 2020-01-15:
@michaelld Here it is https://github.com/gnuradio/gnuradio/commit/73b074f121b0ab2ac38336916a60891c4d88d2cb.

devnulling, 2020-01-23:
@ValZapod It looks like this issue can be closed out with the merger of #3083 , if thats not the case, please reopen this issue.

closed 2020-01-23 by devnulling