gnuradio/gnuradio #3080: UHD USRP Sink GRC generator error

jdemel, 2020-01-15:
I'm at commit 96a5e82123ddc8a3ea6ced7bb903cafacd5113fb

If I open `gr-uhd/examples/grc/uhd_const_wave.grc` I can't generate the flowgraph.

This does only happen with the UHD USRP Sink. The error message is append here. Though, I cut off a lot of the string that's part of the error message.
```
Generating: '/home/johannes/prefix/gnuradio/src/gnuradio/gr-uhd/examples/grc/uhd_const_wave.py'
Generate Error: (TypeError("'NoneType' object is not callable",), 'uhd.usrp_sink(\n    ",".join((${dev_addr},
[...snip ...]

>>> Failure
Traceback (most recent call last):
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 75, in render
    return template.render(**namespace)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7eff9aeea630", line 460, in render_body
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/gui/Application.py", line 716, in _handle_action
    generator.write()
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 83, in write
    for filename, data in self._build_python_code_from_template():
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 127, in _build_python_code_from_template
    blocks=self._blocks(),
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 197, in _blocks
    blocks = expr_utils.sort_objects(blocks, operator.attrgetter('name'), _get_block_sort_text)
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 74, in sort_objects
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/utils/expr_utils.py", line 74, in <dictcomp>
    id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 184, in _get_block_sort_text
    code = block.templates.render('make').replace(block.name, ' ')
  File "/home/johannes/prefix/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/blocks/_templates.py", line 77, in render
    raise TemplateError(error, text)
gnuradio.grc.core.errors.TemplateError: (TypeError("'NoneType' object is not callable",), 'uhd.usrp_sink(\n 

[snip]
```

michaelld, 2020-01-15:
Can you move to the next commit ( https://github.com/gnuradio/gnuradio/commit/cfdef8e6dffb0537918aa91110e86e0b81056eae ) and see if it fixes this issue? This commit is supposed to fix the error you note; here is the issue tracker < https://github.com/gnuradio/gnuradio/issues/3039 >.

jdemel, 2020-01-16:
Yeah. This fixed it. I'm amazed that this bug was fixed so fast. Thanks!

closed 2020-01-16 by jdemel