gnuradio/gnuradio #3085: GNURadio with Python 3 for Windows

rsandler00, 2020-01-17:
Hi, 

Currently, the windows precompiled GNUradio installers found [here ](http://www.gcndevelopment.com/gnuradio/downloads.htm)feature GNURadio 3.8, but only with Python 2.7. My questions are:

1) Is there a currently precompiled Windows version of GNURadio 3.8 w/ Python 3?
2) If there is not, is there a relatively easy way that I could compile one myself?
3) If also not, how could I interface my current Python 3 codebase w/ the GNURadio Python 2 output?

Thanks!

alexcolpitts96, 2020-01-21:
@rsandler00 I have previously interfaced python3 with GR 3.7 (python2) using sockets, saving data in small chunks for processing (I used tagged file sinks), or collecting all the data post processing it.

rsandler00, 2020-01-22:
Thanks @alexcolpitts96. I am hoping I do not have to go that route. Has anyone tried to compile with Python 3 for Windows 10?

marcusmueller, 2020-02-13:
This is really a "borderline" issue: kind of a feature/bug report, but then again, mostly a discussion. Since it's been inactive for > 3 weeks, closing it, and inviting everyone to have _questions_ and _discussions_ on Discuss-gnuradio@gnu.org <discuss-gnuradio@gnu.org>

closed 2020-02-13 by marcusmueller