gnuradio/gnuradio #3091: GRC disabling a block remove it and its connections

ChristopheSeguinot, 2020-01-17:
Hi, 
This bug affect GRC 3.9 (not tested for GRC 3.8)
When a block was disabled in older GRC, this block and its connections were greyed. 
At present the block is completely removed (we need ctrl Z to revert deletion)
Is this a bug ? 

Regards, Christophe

ekigwana, 2020-01-17:
I have noticed this behavior occasionally as well.

haakov, 2020-01-26:
Certainly sounds like a bug! I can't say I've run into this problem myself though.

Are you able to reproduce this behaviour consistently? If so, can you share .grc file? :)

ChristopheSeguinot, 2020-01-27:
Hi, 
I can confirm that this bug occurs on both 3.8 and 3.9 versions running under Ubuntu 18.04. It occurs on all my .grc files like the one attached. Every time I disable a block, it is removed from the graph and its connections as well. I never succeeded in correctly disabling a block.
Question: is this issue only related to Ubuntu or Linux? 
I can provide more information on my config if needed.  
Regards , Christophe 

[test_disabling_a_block.py.zip](https://github.com/gnuradio/gnuradio/files/4116129/test_disabling_a_block.py.zip)

ChristopheSeguinot, 2020-01-31:
I discovered that this bug: 

- does not affect gnuradio 3.8 obtained from debian repository (https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation)
- affect GNURadio 3.8 and 3.9 build from source
did not had time to investigate further...

velichkov, 2020-02-05:
Hi @ChristopheSeguinot,

> It occurs on all my .grc files like the one attached.
> test_disabling_a_block.py.zip

The attached archive contains only the python file. Can you attach one of the grc files that you've used to reproduce this issue?

I'm not able to reproduce it with both 3.8 and 3.9 compiled from source on Fedora 29 with QT 5.11.3. I've tested with several of the example flowgraphs in the gnuradio's repository, disabling several blocks and it works like expected without removing the disabled block nor its connections.

ChristopheSeguinot, 2020-02-05:
Hi 
It really seems that this is related to my installation (I was using gnuradio 3.7 before and may be I made a mistake while removing older package ...). Here is the correct grc file. 
[test_disabling_a_block.grc.zip](https://github.com/gnuradio/gnuradio/files/4158281/test_disabling_a_block.grc.zip)

aru31, 2020-02-05:
Hi,
I also face this bug occasionally, on disabling (pressing d) or bypassing (pressing b), the block gets completely removed.

velichkov, 2020-02-05:
Hi @ChristopheSeguinot, @aru31,

It turns out there is a View -> "Hide Disabled Blocks" (Ctrl + D) setting. This setting has been added in gnuradio 3.7.5 (ab10b5bc162)

Can you check whether this is enabled on your system and try disabling it?

aru31, 2020-02-05:
Hi @velichkov 
You are right, it was because of the "Hide Disabled Blocks" setting.
Thanks. :)

closed 2020-02-06 by ChristopheSeguinot