gnuradio/gnuradio #3098: gnuradio cmake Configuring incomplete, errors occurred!

hanyelei, 2020-01-19:
Configuring sphinx support...
--   Dependency SPHINX_FOUND = TRUE
--   Enabling sphinx support.
--   Override with -DENABLE_SPHINX=ON/OFF
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Could NOT find GMP (missing: GMPXX_LIBRARY GMP_LIBRARY GMP_INCLUDE_DIR) 
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find MPLIB (missing: MPLIBXX_LIBRARY MPLIB_LIBRARY
  MPLIB_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindMPLIB.cmake:26 (find_package_handle_standard_args)
  gnuradio-runtime/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!

hanyelei, 2020-01-19:
Who will help me?

drmpeg, 2020-01-19:
sudo apt-get install libgmp3-dev

devnulling, 2020-01-19:
I'm closing this issue as it is not a bug.Questions such as this are better served either on the mailing list [0], on Slack (register here [1] and login here [2]), or on IRC [3].

[0] - https://wiki.gnuradio.org/index.php/MailingLists
[1] - https://slack.gnuradio.org/
[2] - https://gnuradio.slack.com/
[3] - https://wiki.gnuradio.org/index.php/IRC

closed 2020-01-19 by devnulling

mcan06, 2021-02-09:
Why did you close this topic, the problem has still been  continue on Ubuntu 20.04 and GNU radio 5.9.2?

adeeb10abbas, 2021-02-10:
@mcan06 I second him. I am stuck here as well....

mcan06, 2021-02-10:
> @mcan06 I second him. I am stuck here as well....

I solved these problems by applying gnu radio chatroom with the help of people within there.

adeeb10abbas, 2021-02-22:
I can't seem to be able to join it. Is there a way you could email me screenshots or something? My email - aa4239@drexel.edu

Dygear, 2021-05-18:
> > @mcan06 I second him. I am stuck here as well....
> 
> I solved these problems by applying gnu radio chatroom with the help of people within there. You can find my answers and questions in there.

It would be far more helpful if you posted the solution here ... because this is now the top search result for this on google.

mcan06, 2021-05-18:
I have a problem when I installing a gr-radar, where do you have a problem?

Dygear, 2021-07-13:
I managed to compile and install it.

```
git clone https://github.com/wbhart/mpir.git ~/mpir
cd ~/mpir
sudo apt install yasm
bash autogen.sh
bash configure.yasm
make
sudo make install
```

drmpeg, 2021-07-13:
@Dygear You need to build the C++ library. Instead of ```config.yasm```, it should be ```./configure --enable-cxx```.

Dygear, 2021-07-14:
So. I can’t for some reason. It’s saying configure is not there. (For mpir)

drmpeg, 2021-07-14:
```configure.yasm``` just executes ```./configure```, so it has to be there.