gnuradio/gnuradio #3099: GRC 'ModuleNotFoundError'

duggabe, 2020-01-19:
After loading GR maint-3.8 from source, I tried to execute gnuradio-companion and got a window titled 'ModuleNotFoundError'. See https://wiki.gnuradio.org/index.php/File:ModNotFound.png
After discussions on #gnuradio and experimentation, I found the following solution:

put the following in your <code>.bashrc</code> or <code>.bash_aliases</code> file and restart your terminal.<br>
For Debian / Ubuntu, and derivatives, and other 32-bit OS use:
<pre>
export PYTHONPATH=/usr/local/lib/python3/dist-packages:/usr/local/lib/python3/site-packages
</pre>
Other 64-bit systems may require:
<pre>
export PYTHONPATH=/usr/local/lib64/python3/site-packages
</pre>

The 'ModuleNotFoundError' window needs to be updated:
1) remove reference to "library path environment variable" (it doesn't seem to be needed).
2) add the information given above for PYTHONPATH.

michaelld, 2020-01-20:
I'd advise you, in any such change, to do:
```
export PYTHONPATH=[STUFF]:$PYTHONPATH
```
just in case the `PYTHONPATH` is set externally somewhere / somehow. It's also just good practice in general to prepend, not overwrite, environment variables ... not -always-, just generally.

duggabe, 2020-01-20:
yes, I will fix that right away in https://wiki.gnuradio.org/index.php/InstallingGR#To_install_system_wide .
If you tell me where to find the code for that error message, I'll look at changing it myself.

velichkov, 2020-01-20:
https://github.com/gnuradio/gnuradio/blob/63e490434a19b757baaa0920055cde94b9362d18/grc/scripts/gnuradio-companion#L24-L34
and
https://github.com/gnuradio/gnuradio/blob/63e490434a19b757baaa0920055cde94b9362d18/grc/scripts/grcc#L23-L33

rear1019, 2020-01-20:
> The 'ModuleNotFoundError' window needs to be updated:
> 
>     1. remove reference to "library path environment variable" (it doesn't seem to be needed).

It is required if a non-standard ``CMAKE_INSTALL_PATH`` is used, i.e. anything other than ``/usr`` or ``/usr/local`` (depending on the distribution?) => Do not remove the reference to ``LD_LIBRARY_PATH``.

duggabe, 2020-01-20:
@rear1019 OK. can you help me with what it should say?

duggabe, 2020-01-21:
I am working on this issue and realize there is more explanation than could reasonable fit on the pop-up. So I was wondering if it would make sense to create a separate wiki page to explain it all, and put a link to that in the 'ModuleNotFoundError' window. What do you think?

777arc, 2020-01-22:
I think that's a great idea, let me know if you want assistance creating that wiki page or editing the popup window to link to it.  If the text generation of the popup is anything like the Documentation tab of a block, we won't be able to do a clickable link, but we can have the URL people can copy/paste.

duggabe, 2020-01-23:
There are three places where the 'ModuleNotFoundError' is generated:
1) executing `gnuradio-companion`
2) executing `grcc MorseGen.grc` (for example)
3) executing `python3 MorseGen.py` (for example)

The first case produces a graphic window with the error message. The link is not clickable.
The second case produces the text of the error message on the terminal. The link is clickable.
I have modified both cases 1 and 2 to look like this:
```
ModuleNotFoundError

Cannot import gnuradio.

Is the Python path environment variable set correctly?
    All OS: PYTHONPATH

Is the library path environment variable set correctly?
    Linux and MacOSX: LD_LIBRARY_PATH
    Windows: PATH

See https://wiki.gnuradio.org/index.php/ModuleNotFoundError

(No module named 'gnuradio')
```
The third case produces a condensed version of the error message:
```
~/gr-morse-code-gen $ python3 MorseGen.py
Traceback (most recent call last):
  File "MorseGen.py", line 26, in <module>
    from gnuradio import qtgui
ModuleNotFoundError: No module named 'gnuradio'
```
I don't know how to modify case (3).

FYI, I am testing on my fork of master 3.9.0.0-git

777arc, 2020-01-23:
You could just replace "from gnuradio import qtgui" with:

try:
    from gnuradio import qtgui
except ModuleNotFoundError:
    sys.exit("Python could not find the gnuradio module, check your PYTHONPATH, for help setting your path refer to this page <url of wiki page>")

just make sure "import sys" happens before that block of code

rear1019, 2020-01-23:
> I am working on this issue and realize there is more explanation than could reasonably fit on the pop-up. So I was wondering if it would make sense to create a separate wiki page to explain it all, and put a link to that in the 'ModuleNotFoundError' window. What do you think?

This is indeed the way to go in my option.

> There are three places where the 'ModuleNotFoundError' is generated:
> […]
> The second case produces the text of the error message on the terminal. The link is clickable.

Whether the link is clickable or not depends on the terminal used. But at least there is a link which can be selected and copied.

> I have modified both cases 1 and 2 to look like this:
>
> ```
> ModuleNotFoundError
>
> Cannot import gnuradio.
>
> Is the Python path environment variable set correctly?
>     All OS: PYTHONPATH
>
> Is the library [...]

Looks fine to me.

> The third case produces a condensed version of the error message:
>
> ```
> ~/gr-morse-code-gen $ python3 MorseGen.py
> Traceback (most recent call last):
>   File "MorseGen.py", line 26, in <module> [...]

Fixing this would require changing the GRC compiler responsible for generating Python code out of a flowgraph. The following check must be added to each Python script generated:
https://github.com/gnuradio/gnuradio/blob/076d264d2443efbdf2a1f058da701309af9dc304/grc/scripts/gnuradio-companion#L72-L75
Replace ``die()`` by ``print(error_message)`` and exit afterwards.

The wiki page looks fine to me. But there is a problem the way the prefix is retrieved: The listed method will only work if the binary ``gnuradio-config-info`` can be found in ``PATH``. Moreover, as ``gnuradio-config-info`` links against ``libvolk``, ``libgnuradio-runtime`` and ``libgnuradio-pmt``, the binary will only work if these libraries are found. Which means that a correctly set ``LD_LIBRARY_PATH`` might be required (depending on the distribution and the used install prefix).

duggabe, 2020-01-25:
Fixes: https://github.com/gnuradio/gnuradio/pull/3119

michaelld, 2020-01-25:
Merged #3119 ... closing.

closed 2020-01-25 by michaelld