gnuradio/gnuradio #3100: Filter design tool not opening

devnulling, 2020-01-19:
Testing with latest master , ubuntu 18.04. When trying to open, nothing happens and get this error in the terminal:

```
/usr/lib/python3.6/subprocess.py:616: ResourceWarning: unclosed file <_io.BufferedReader name=11>
  _cleanup()
/usr/lib/python3.6/subprocess.py:786: ResourceWarning: subprocess 22460 is still running
  ResourceWarning, source=self)
```

Semi related, but with older filter tool design: https://github.com/gnuradio/gnuradio/issues/1518

cbdonohue, 2020-01-28:
I was able to open the filter design tool with Ubuntu 19.10 and master commit  4846be906ad2d4b9d60dfae07f3c4dc6dd95fa95

However, pyqtgraph 0.11.0.dev0 didn't work, and I had to install pyqtgraph 0.11.0.dev0+g6a2791e
`pip3 install git+https://github.com/pyqtgraph/pyqtgraph@develop`

wbarnha, 2020-03-02:
Tried the same solution for 3.8 on Ubuntu 18.04 with pyqtgraph, no luck running it directly in GNU Radio. I tried running filter_design.py as indicated from #1518 with no luck. The issue is that in `filter_design.py`, (see line 40), the block of code is
```
try:
    from numpy.fft import fftpack as fft_detail
except ImportError:
```
Changing it to 
```
try:
    from scipy import fft as fft_detail
except ImportError:
```
gets filter_design.py working when the program is run from command line.

wbarnha, 2020-03-07:
I had to setup a new development environment and it looks like that specific version of PyQt is also needed to get the Filter Tool GUI working again as well as changing the modules imported by Python.

nickoe, 2020-03-30:
I think this is addressed via this PR on maint-3.8 https://github.com/gnuradio/gnuradio/pull/3259 and https://github.com/gnuradio/gnuradio/pull/3291 on master.

nickoe, 2020-03-30:
With https://github.com/gnuradio/gnuradio/pull/3291 (master) and https://github.com/gnuradio/gnuradio/pull/3259 (maint-3.8) merged, which essentially included the proposal to use `numpy.fft` directly https://github.com/gnuradio/gnuradio/pull/3291/commits/d3cffa3580a5377c48c313d628c298c81cee7573#diff-aa15db7a96c16ff668fd14aa585dfb90R29 I think we can close this one.

closed 2020-03-30 by nickoe