gnuradio/gnuradio #3106: UHD: WBFM examples needs to be fixed up

devnulling, 2020-01-22:
SyntaxError: unexpected EOF while parsing
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block audio_output
Traceback (most recent call last):
  File "/home/user/sdr/installs/v3.15.0.0_master/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

Loading: "/home/user/sdr/installs/v3.15.0.0_master/share/gnuradio/examples/uhd/uhd_wbfm_receive.grc"
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate parameter block audio_output
Traceback (most recent call last):
  File "/home/user/sdr/installs/v3.15.0.0_master/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 216, in renew_namespace
    value = eval(parameter_block.params['value'].to_code(), namespace)
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

velichkov, 2020-01-23:
The `Value` in address and audio_output blocks needs to be a string, similar fixes are needed for some of the other uhd examples as well.
```diff
diff --git a/gr-uhd/examples/grc/uhd_wbfm_receive.grc b/gr-uhd/examples/grc/uhd_wbfm_receive.grc
index 20979d36b4..34a1d21780 100644
--- a/gr-uhd/examples/grc/uhd_wbfm_receive.grc
+++ b/gr-uhd/examples/grc/uhd_wbfm_receive.grc
@@ -137,7 +137,7 @@ blocks:
     label: IP Address
     short_id: a
     type: ''
-    value: addr=192.168.10.2
+    value: '"addr=192.168.10.2"'
   states:
     bus_sink: false
     bus_source: false
@@ -171,7 +171,7 @@ blocks:
     label: Audio Output Device
     short_id: O
     type: ''
-    value: ''
+    value: '""'
   states:
     bus_sink: false
     bus_source: false
```

devnulling, 2020-01-23:
@velichkov Yeah i had identified the string issue before making this post and just put it up as a reminder to myself. However, after fixing the string issue, this parameter block issue came up https://github.com/gnuradio/gnuradio/issues/3107

There is some other changes I intend to make for the WBFM example as well.

closed 2020-02-09 by marcusmueller