gnuradio/gnuradio #3116: Compiling Gnuradio 3.8 with gcc 9.2.1 and g++ 9.2.1

sirjansel0t, 2020-01-24:
Hello,

I was able to compile gnuradio 3.8 with g++ and gcc 9.2.1 however, one of the make tests failed:

The following tests FAILED:
         25 - qa_volk_32f_8u_polarbutterflypuppet_32f (Failed)
Errors while running CTest

I then rant ctest:

$ ctest -V -R qa_volk_32f_8u_polarbutterflypuppet_32f

UpdateCTestConfiguration  from :/home/jjjj/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/jjjj/gnuradio/build/DartConfiguration.tcl
Test project /home/jjjj/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 25
    Start 25: qa_volk_32f_8u_polarbutterflypuppet_32f

25: Test command: /bin/sh "/home/jjjj/gnuradio/build/volk/lib/volk_32f_8u_polarbutterflypuppet_32f_test.sh" "/home/jjjj/gnuradio/build/volk/lib"
25: Test timeout computed to be: 10000000
25: RUN_VOLK_TESTS: volk_32f_8u_polarbutterflypuppet_32f(131071,1)
25: generic completed in 2.82904 ms
25: u_avx completed in 0.411041 ms
25: u_avx2 completed in 0.413137 ms
25: offset 8 in1: 2 in2: 0.498039 tolerance was: 1e-06
25: offset 9 in1: -4 in2: -2.49804 tolerance was: 1e-06
25: offset 10 in1: -4 in2: -0.996078 tolerance was: 1e-06
25: offset 11 in1: -8 in2: -4.99608 tolerance was: 1e-06
25: offset 12 in1: -4 in2: -0.996078 tolerance was: 1e-06
25: offset 13 in1: 8 in2: 4.99608 tolerance was: 1e-06
25: offset 14 in1: -8 in2: -1.99216 tolerance was: 1e-06
25: offset 15 in1: 16 in2: 9.99216 tolerance was: 1e-06
25: offset 16 in1: 2 in2: 0 tolerance was: 1e-06
25: offset 17 in1: 4 in2: 0 tolerance was: 1e-06
25: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx2
25: offset 16 in1: 0 in2: 1 tolerance was: 0
25: offset 17 in1: 0 in2: 1 tolerance was: 0
25: offset 18 in1: 1 in2: 0 tolerance was: 0
25: offset 19 in1: 1 in2: 0 tolerance was: 0
25: offset 25 in1: 0 in2: 1 tolerance was: 0
25: offset 26 in1: 0 in2: 1 tolerance was: 0
25: offset 28 in1: 1 in2: 0 tolerance was: 0
25: offset 29 in1: 1 in2: 0 tolerance was: 0
25: offset 34 in1: 0 in2: 1 tolerance was: 0
25: offset 36 in1: 0 in2: 1 tolerance was: 0
25: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx2
25: Best aligned arch: u_avx
25: Best unaligned arch: u_avx
1/1 Test #25: qa_volk_32f_8u_polarbutterflypuppet_32f ...***Failed    0.01 sec

Any ideas why this test fails?  It works when I use gcc/g++ 8.3.  Also what modules rely on this?  Thank you.

dkozel, 2020-01-30:
Hi @sirjansel0t Thanks for the report, though it should go into the VOLK bug tracker. I've made a short issue there. Can you please continue the report there?

That VOLK kernel is not used in GNU Radio at this point that I can find. If you encounter any related problem with GNU Radio then please do reopen this issue!

closed 2020-01-30 by dkozel

jdemel, 2020-02-08:
The kernel that is tested here is `volk_32f_u8_polarbutterfly_32f` which is used in `polar_decoder`. We continue this discussion in gnuradio/volk#340.