gnuradio/gnuradio #3117: qtgui swig error in gnu3.8-maint with CentOS 7.7

mwk088, 2020-01-24:
OS:  CentOS 7.7
GNU:  maint-3.8

Due to the error with the QWT 6.1.1 package error associated with CentOS 7, I need to build QWT from source.  I build it version 6.1.3, tell GNU where to find it, and cmake finds it and enables the qtgui module, but when I created a flowgraph in GRC, I get the error seen below.

cmake command
` cmake3 -DENABLE_INTERNAL_VOLK=OFF -DCMAKE_INSTALL_PREFIX=${PB_PREFIXDIR} -DPYTHON_LIBRARY=/usr/lib64/libpython3.6m.so -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DENABLE_GR_LOG=OFF $config_opt -DQWT_INCLUDE_DIRS=/usr/local/qwt-6.1.3/include -DQWT_LIBRARIES=/usr/local/qwt-6.1.3/lib/libqwt.so ../
`
Error seen trying to run a flowgraph with a QTGui block
`Traceback (most recent call last):
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libqwt.so.6: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/__init__.py", line 32, in <module>
    from .qtgui_swig import *
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 17, in <module>
    _qtgui_swig = swig_import_helper()
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_qtgui_swig')
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_qtgui_swig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libqwt.so.6: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/truearrow/6.3/testy.py", line 24, in <module>
    from gnuradio import qtgui
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/__init__.py", line 36, in <module>
    from .qtgui_swig import *
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 17, in <module>
    _qtgui_swig = swig_import_helper()
  File "/opt/truearrow/6.3/lib64/python3.6/site-packages/gnuradio/qtgui/qtgui_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_qtgui_swig')
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_qtgui_swig'

>>> Done (return code 1)`

Any ideas?

Thank you
Mark

rear1019, 2020-01-24:
> ```
> […]
> ImportError: libqwt.so.6: cannot open shared object file: No such file or directory
> […]
> ```

``ldconfig`` run and/or ``LD_LIBRARY_PATH`` set (depending on distribution)?

mwk088, 2020-01-24:
I fixed it....I had to uninstall the canned qwt packages on Centos 7, which are only at 5.9.2.  I then dd the following:  

`wget ftp://ftp.pbone.net/mirror/ftp5.gwdg.de/pub/opensuse/repositories/home:/maxiotis:/ir2/CentOS_7/x86_64/qwt-6.1.3-2.33.x86_64.rpm
wget ftp://ftp.pbone.net/mirror/ftp5.gwdg.de/pub/opensuse/repositories/home:/maxiotis:/ir2/CentOS_7/x86_64/qwt-qt5-devel-6.1.3-2.33.x86_64.rpm
wget ftp://ftp.pbone.net/mirror/ftp5.gwdg.de/pub/opensuse/repositories/home:/maxiotis:/ir2/CentOS_7/x86_64/qwt-qt5-6.1.3-2.33.x86_64.rpm
yum -y install qwt*`

closed 2020-01-24 by mwk088

Amalinda, 2020-05-22:
Just a note: on Arch, pacman -S qwt solves this issue on a fresh install. qwt does not install as a dependency by default.