gnuradio/gnuradio #3120: clang-format option `SortIncludes` potentially breaks code

sbmueller, 2020-01-25:
The `SortIncludes` option in the .clang-format file potentially breaks code. Includes of each group (e.g. `^[a-Z]*>$`) are sorted alphabetically. This is dangerous since the include order should be determined by functionality, not by name.

Example what this could break: Qt uses the keyword `slots`. However, Python also uses this name for other purposes [1]. This makes it necessary to include Python before Qt, or else the `slots` statements in Python will be treated as Qt keywoards, which leads to build failure.
I reckon only by coincidence, this has not yet caused errors in gr-qtgui.

One solution could be to disable the `SortIncludes` option, which is highly controversial [2].

[1] https://stackoverflow.com/questions/23068700/embedding-python3-in-qt-5
[2] https://news.ycombinator.com/item?id=21580829

ValZapod, 2020-01-25:
@sbmueller It could be even worse. https://travisdowns.github.io/blog/2019/11/19/toupper.html

sbmueller, 2020-01-28:
To add some more thought to this: Disabling this option won't result in different `clang-format` outputs since include order will remain untouched. All formatted source files should remain unmodified when re-running `clang-format`.

sbmueller, 2020-01-28:
Related #1664

sbmueller, 2020-03-05:
Won't be changed, discussion can be found in #3238 .

closed 2020-03-05 by sbmueller