gnuradio/gnuradio #3125: Code cleanup

ValZapod, 2020-01-25:
https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/hier_block2_detail.cc#L58-L59

std::min, maybe?

https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/local_sighandler.cc#L58
You cannot throw in destructor. Use try..catch See https://www.viva64.com/en/w/v509/

https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/logger.cc#L230
Why there is two functions with the same name? Is this some magic?

https://github.com/gnuradio/gnuradio/blob/0256f9ed955ef293a1a8a36e3430d98ac7edd46c/gr-digital/lib/ofdm_cyclic_prefixer_impl.cc#L199
This is undefined behavior.

michaelld, 2020-01-26:
> https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/hier_block2_detail.cc#L58-L59
> 
> std::min, maybe?

Thinking there's a reason why that's done, but I can't recall off the top of my head.

> https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/local_sighandler.cc#L58
> 
> You cannot throw in destructor. Use try..catch See https://www.viva64.com/en/w/v509/

Indeed!

>
https://github.com/gnuradio/gnuradio/blob/192307320d2482957b73af74d67303ea4a19d60d/gnuradio-runtime/lib/logger.cc#L230
> 
> Why there are two functions with the same name? Is this like some magic?

You mean `logger_get_level`? Different prototypes, same name. C++ magic!

> https://github.com/gnuradio/gnuradio/blob/0256f9ed955ef293a1a8a36e3430d98ac7edd46c/gr-digital/lib/ofdm_cyclic_prefixer_impl.cc#L199
> 
> This is undefined behavior.

In what sense? The `++` will have 1st priority, then `.size()`, then `%`. What's not to like here?

michaelld, 2020-01-26:
On that last item. I do actually think it will work because of operator priority. That said, I think it can  be rewritten to be less concise but clearer & optimized to be just as good.

ValZapod, 2020-01-26:
@michaelld It will work of course in gcc/clang. But this is still UB. You cannot change d_state twice in one line... https://www.viva64.com/en/w/v567/ It became defined only in C++ 17, if I remember correctly the corner cases.

```
++d_state;
d_state %= d_cp_lengths.size();
```

ValZapod, 2020-01-27:
Okay. It may be correct but you can do `d_max_output_buffer = d_min_output_buffer = std::vector<size_t>(std::max(max_outputs, 1), 0);`, right?

michaelld, 2020-01-29:
> Okay. It may be correct but you can do `d_max_output_buffer = d_min_output_buffer = std::vector<size_t>(std::max(max_outputs, 1), 0);`, right?

Yes one could do that. I don't think it helps the code though. This is done in the constructor, and so executed just once as the hier block is being created ... so, there's no real performance gain by this suggested change ... and, if for some reason they should be different then we'd need to split this into 2 lines anyway. Please just leave it be at this time. We can always revisit down the road!

ValZapod, 2020-01-29:
@michaelld Heh, you just asked to change to one variable in 3 places)) And this is std::max called twice...

michaelld, 2020-01-29:
Indeed! That's the life of a reviewer and submitter ... neither of us know what to expect day to day! Might be one thing one day, and another thing another day. I try to be as consistent as possible, of course, but I am leveraging my expertise while still learning! Some changes make more sense, in general as well as "today" ... but maybe not in other contexts or "tomorrow" ... ah, the life!

marcusmueller, 2020-04-12:
> std::min, maybe?

no,  max is what's needed here.

> You cannot throw in destructor.

That's not correct. You mustn't throw in a `noexcept` context, but watch the signature of the destructor closely!

This is the local signal handler's dtor. It throwing is intended behaviour.

> Why there are two functions with the same name? Is this like some magic? 

This is function overloading, one of the most basic C++ features.

closed 2020-04-12 by marcusmueller