gnuradio/gnuradio #3134: grc cannot build a flowgraph any longer

dl1ksv, 2020-01-28:
grc builds no flowgrap. It fails with

Generate Error: 'catch_exceptions'
>>> Failure
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3/dist-packages/gnuradio/grc/gui/Application.py", line 705, in _handle_action
    generator.write()
  File "/usr/local/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 83, in write
    for filename, data in self._build_python_code_from_template():
  File "/usr/local/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 122, in _build_python_code_from_template
    'catch_exceptions': fg.get_option('catch_exceptions')
  File "/usr/local/gnuradio/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 141, in get_option
    return self.options_block.params[key].get_evaluated()
KeyError: 'catch_exceptions'

Indeed the options block has no parameter named catch_exceptions.

haakov, 2020-01-28:
Try deleting this file:
`.cache/grc_gnuradio/cache.json`

devnulling, 2020-01-28:
What version/commit of GR are you using?

dl1ksv, 2020-01-28:
Deleting the cache file does not change anything.
I'm using the latest commit from the master branch from today that is

3ab13921919b

velichkov, 2020-01-28:
Hi @dl1ksv,

> Indeed the options block has no parameter named catch_exceptions.

It have `catch_exception` parameter.

https://github.com/gnuradio/gnuradio/blob/785a889ae877c5e4bb6086b3742a3ab13921919b/grc/blocks/options.block.yml#L109-L116

Make sure that GRC is reading blocks definitions from the correct place in case you are having multiple versions installed at the same time.  In 65a8c6c526d05ce5bc060725d9338e81a4bbbc22 the type was changed from bool to enum.

> I'm using the latest commit from the master branch from today that is
> 
> 3ab13921919b

You need to copy the first few (5-7) 785a889 chars from the sha1 and not the last. 785a889

dl1ksv, 2020-02-04:
Though I did a fresh install of gnuradio the 
~/.gnuradio/config.conf file pointed to an older still existing global_block_path.
No idea why this path was not modified by the fresh installation.
Due to the wrong path deleting cache.json had no effect as it was rebuild from the wrong location.

Correcting the global_block_path and deleting cache.json did it.

closed 2020-02-04 by dl1ksv