gnuradio/gnuradio #3140: backslashes in Windows pathnames not escaped when selecting a QSS theme

wahspilihp, 2020-01-30:
If you select a QSS theme via the options dialog, the backslashes in the path are neither escaped nor converted to forward slashes. Instead the string is inserted in the template (at line 351 of flow_graph.py.mako) as a standard (not raw) string. This means the filename is incorrect when the string contains, for example, the `\t` of  `...\gnuradio\themes\...`.

This affects at least the current 3.8 windows binary version, and none of the previously closed bug are since 3.8 came out, but I have not tested any other version.

marcusmueller, 2024-02-14:
Can't reproduce right now, will ask the intarwebs

wahspilihp, 2024-02-16:
I’m no longer involved in the project I was working on then and I’m not doing any GNuradio work on windows, but it doesn’t look like setStyleSheetFromFile or the calling code (now on like 341) has been updated. 

I’ve no idea though  if the escaping is added somewhere else, which would probably be more appropriate — I suspect the bug is actually in the whatever calls the file picker, so it might well appear elsewhere too.