gnuradio/gnuradio #3141: Print function used without __future__ import

wahspilihp, 2020-01-30:
The print function (with keyword argument) is used in [flow_graph.py.mako](https://github.com/gnuradio/gnuradio/blob/maint-3.8/grc/core/generator/flow_graph.py.mako#L250) without `from __future__ import print_function`

Inserting the import after line 8 appears to work: a quick skim of the template did not show anywhere the print statement was used to print a tuple or where the parenthesis were missing (and if there is, it would need fixing for 3.9 anyway), but I have not thoroughly tested the fix. Users cannot simply use an import block because future imports must come before any others.

marcusmueller, 2020-02-13:
Hi @wahspilihp, thank you for the bug report!

Yeah, if parentheses had been missing (i.e. old-school Python2 `print "foobar"`), then it would have failed on Python3 (which is probably the reason this slipped by us: most GR devs use Python3 with GNU Radio 3.8.x.x by now, which is the only release series with both Py2 and Py3 support). 

We need to fix this! Do you think you have the bandwidth to really put that into a minimal pull request against the `maint-3.8` branch?

nickoe, 2020-03-31:
@wahspilihp are you able follow-up on this one?

willcode, 2021-03-03:
Addressed by PR above, closing.

PR #3442 (2020-05-21): grc: Updated flow_graph.py.mako with import statement (maint-3.8)
Fixes #3141 
Added:
 ~~~~
from __future__ import print_function
PR #3508 (2020-05-24): grc: fix broken flow_graph template
Fixes #3506.
Fixes #3141.

The change in #3442 put the `from __future__ import print_function` into a Mako Python block instead of the generated flowgraph. This broke GRC, at least on Python 3.

To fix this I reverted #3442 and moved `from __future__ import print_function` to the correct place. I also put it in a Mako `% if` block so it's only generated in Python 2.
closed 2021-03-03 by willcode