gnuradio/gnuradio #3145: Ctrl+0 should reset zoom to 100%

primercuervo, 2020-01-30:
Now that we just were mindblown with the realization that you can, indeed, zoom in and out a flowgraph using `ctrl`+mouse wheel, it would be useful to have a hotkey to reset the zoom to 100%.

dkozel, 2020-01-30:
`ctrl`+`+` and `ctrl`+`-` should zoom in and out.

skoslowski, 2020-01-30:
Haha, that is hilarious :smile:

dkozel, 2020-01-31:
The mindblowing has continued. First it was about half of the hackfest participants, then it was more folks over dinner at FOSDEM. One pulled out a laptop on the spot to try it. ;) Great to have people smile.

jaskiratsingh2000, 2021-07-17:
Hey @primercuervo @skoslowski @dkozel I am quite beginner and super interested to begin my contribution towards this repo and this issue seems interesting to me. Can you please let me know that where exactly do I have to make changes or add functionalities?
Any examples corresponding to this would be highly appreciable. Thanks!

jaskiratsingh2000, 2021-07-17:
Also, is this fully opened yet @mbr0wn ?