gnuradio/gnuradio #3155: DOI for releases

noc0lour, 2020-02-01:
Add documentation how and which DOI to use if citing gnuradio in published documents.

777arc, 2020-02-09:
Wasn't aware there was one.  Were you thinking of Blossom's 2009 paper "Exploring gnu radio" paper with 88 citations that's only available in PS format?  I don't even see a DOI associated with it, seems self-published. 

Update- just found "GNU radio: tools for exploring the radio frequency spectrum" from 2004, with 380 citations according to Google Scholar.  It's not a paper, but an article from linuxjournal.com- https://www.linuxjournal.com/article/7319.  That's probably why it did not show up at all in any of my "gnuradio" or "gnu radio" searches...  And it only shows up as a citation, and once again I don't think there's a DOI associated with it. 

Unless I'm missing something, it seems like the best thing we can do is just keep citing the one with 380 citations, and you do so by just copying the citation other papers have used, since there's no DOI, but Google Scholar will connect them if they have the same title/author/year.

dkozel, 2020-02-09:
We can generate DOIs for specific releases and add examples of how to

777arc, 2020-02-09:
Usually when a paper cites software, they are really just linking to that software's URL and pointing out further resources about it.  So you might have a paper that says "We used GNU Radio [1] to do blah blah blah" and then at the bottom there is a 

[1] GNU Radio, version 3.8.0. Available: https://www.gnuradio.org/

You could add an official DOI to that, but the reader isn't going to care, and if the DOI changes every release then each one will only have a few citations and it will never show up on searches.  Seems like if we want to have a DOI, it should be 1 and never change, so that at least it accumulates citations.

I think the bigger problem to address is that the 380 citation article doesn't show up on Google Scholar, i.e. it's not indexed by it, and only shows up as a citation if you specifically search the title.  If it was indexed by Google Scholar it would probably be the first thing that shows up when you search "GNU Radio" and then people would be more likely to cite it as an alternative to citing software directly.

noc0lour, 2020-02-10:
I think a DOI is useful to refer to a specific version of GNU Radio which would be useful for reproduction purposes. So if you cite GNU Radio 3.8 with this DOI if someone reading (the digital version) of your paper follows the link he finds with which specific version you did produce these results. Since we might have bugs and or secret features which might not be there in future or earlier releases. In theory someone tried to set up https://zenodo.org/ to create DOIs but the webhook never fired (yet), probably some configuration error.

noc0lour, 2020-02-10:
This has been automagically created for GNU Radio 3.8 https://zenodo.org/record/3364804 
Probably the hook works nonetheless?

jdemel, 2022-03-22:
I'd really like to see polished GR Zenodo releases. 

The latest Zenodo entry is https://zenodo.org/record/5918388

I suggest to polish the Zenodo DOI with a `.zenodo.json` File in the GR top-level.

I added such a file to VOLK. Thus, [the latest Zenodo entry](https://zenodo.org/record/6052858) is smth I consider useful to be cited. This has benefits for GR contributors in academia because you can add your ORCID as well.

I suggest to add all GR contributors as authors. I know the list would be lengthy but that'd be reasonable. Besides, I'd suggest to sort the authors alphabetically with 2 exceptions:
1. put the current maintainer names first. (Josh and Jeff).
2. Contributors with only github handles go last.

In order to make this a painless procedure, I suggest to adopt the [scripts we have in VOLK](https://github.com/gnuradio/volk/blob/main/scripts/tools/update_citations.sh). The GR scripts should go to [dtools/bin](https://github.com/gnuradio/gnuradio/tree/main/dtools/bin). Obviously, the `.zenodo.json` needs to reside in the root directory. The initial addition would probably require some manual work for clean-up. Alternate places for these scripts would be `tools` and `gr-utils`. That's probably mainly a maintainer preference.

mormj, 2022-03-22:
I have no objections to this, and I think `tools` would be the right place for the scripts

777arc, 2022-10-06:
Some minor issues I ran into when actually trying to use this to cite GR- 1) the DOI appears to change every release which doesn't seem ideal from a google scholar statistics point of view and 2) it wants to cite it using a list of dozens of authors, look at the "cite as..." part on the bottom-right.  Plus either by mentioning it in the text or using a cite like shown below, you can share which version of GR you were on. 

[1] GNU Radio, version 3.10.2. Available: https://github.com/gnuradio/gnuradio

jdemel, 2022-10-07:
Regarding 1)
Every release receives its own DOI. I'd argue that is strictly necessary because this enables you to cite specific versions.

However, you may use [10.5281/zenodo.2704343](https://doi.org/10.5281/zenodo.2704343) to cite "all versions". Thus, this DOI sends you to the latest GR version at any moment in time.

@willcode If you'd do the 3.9 maint release before the 3.10 release, the latest version would point at the latest 3.10 release most of the time. Just a minor comment though.

Regarding 2)
Personally, I feel like all contributors should be part of the authors list. Even if this is a very long list. You may change the details though.
- truncate the authors list in your citation.
- Let BibTex truncate it for you.
- Remove all authors and cite the DOI only.
- etc.

Personally, I'd like a DOI for every citation because it makes it easier to work with. Also, it integrates well with scientific publications. e.g. a DOI stresses the point that the code is a publication that is just as important as any other paper. I feel like this is not yet universally understood.

willcode, 2022-10-07:
Github shows the latest version as v3.10.4.0. In any case, we'll only have 3.10 releases for the next few quarters. Also, the .zenodo.json file is only in the maint-3.10 branch.

marcusmueller, 2024-01-08:
with the .zenodo-json in place, we actually would need some kind of tooling so that 

1. the release JSON can get updated without manually adding contributors
2. this can happen without much hassle in each release commit

willcode, 2024-01-08:
I run the tool in the gnuradio-metadata repo on maint-3.10 for each release, though not for the recent patch level releases.