gnuradio/gnuradio #3158: Packet tx/rx hier blocks doesn't work properly

mohmmadd, 2020-02-02:
In GNURadio 3.8 (master branch), when I try to run packet_loopback_hier.grc at: gnuradio-master/gr-digital/examples/packet, I don't receive anything. In GNURadio 3.7.13, it works but some packets drop (When I send a text message in simulation, some of the text is lost). No warning message is shown. I think it should be a bug in gnuradio 3.8 (also 3.7).

marcusmueller, 2020-02-13:
I'm confused: The master branch is not GNU Radio 3.8. GNU Radio 3.8 lives on the maint-3.8 branch.

Am I right in assuming this is about the master branch? What does `gnuradio-config-info --builddate --version` say?

mohmmadd, 2020-02-17:
It said 3.9.0.0 git, but today I tested the example flowgraph on gnuradio 3.8.1 and it doesn't work. This example works on 3.7.13.5.

mohmmadd, 2020-04-18:
packet_loopback_hier.grc at: gnuradio-master/gr-digital/examples/packet example doesn't yield any output in GR3.8 and 3.9git, but it yields outputs in GR3.7.

closed 2020-04-22 by mohmmadd