gnuradio/gnuradio #3167: Problem with COPY block in Gnu Rdio 3.8

brandtbg, 2020-02-03:
I can't find a way to control the COPY block in GnuRadio 3.8.
There is a EN input but I can't make it work.
In 3.7 it was possible to put my own string as trigger (for example “variable_qtgui_check_box”) but now I have only switch (ON/OFF) into the block and no way to put a string.
Am I do something wrong or it's a issue?
Thanks in advice!

velichkov, 2020-02-03:
The switch widget for bool typed parameters was added in #2028 and was reverted in the `maser` branch in #2816 but it's still not reverted in the `maint-3.8` branch.

@brandtbg If you've installed gnuradio from source you may try reverting one commit - `git revert 007923c` and then rebuild and install.

A possible workaround is to manually edit the `.grc` file and put the id of gui block in `enabled:` parameter, something like

```yml
- name: blocks_copy_0
  id: blocks_copy
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    enabled: id_button_enabled
```

Also this issue is a duplicate of #2683.

dkozel, 2020-04-08:
Fixed by reverting the UI change.

closed 2020-04-08 by dkozel