gnuradio/gnuradio #3175: Package has no install method: python-click-plugins

mwk088, 2020-02-06:
OS:  CentOS 8
Gnuradio 3.8

commands performed:
dnf install python3-pip
dnf install git
pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs auto-config
pybombs recipes add-defaults
pip3 install click-plugins
pybombs -v -v install gnuradio

I perform the above commands and using pybombs I am unable to install gnuradio.  I keep on getting the error below.

Package has no install method: python-click-plugins

Please advise, thank you

Mark

velichkov, 2020-02-06:
Hi @mwk088,

This could be an issue in [gr-recipes](https://github.com/gnuradio/gr-recipes) or [pybombs](https://github.com/gnuradio/pybombs/) but for sure it's not an issue in gnuradio. Please next open such issues in the correct repository. 

Do you have [EPEL](https://fedoraproject.org/wiki/EPEL) repository installed and enabled on your syste?

https://download.fedoraproject.org/pub/epel/8/Everything/x86_64/Packages/p/python3-click-plugins-1.1.1-4.el8.noarch.rpm

mwk088, 2020-02-06:
I do have everything enabled and I will open in the correct repo.  To be fair, I am trying to build gnuradio, and clicked-plugins is listed as a dependency required to build gnuradio.  

From now on, anything pybombs related will go into that repo.

closed 2020-02-06 by mwk088

velichkov, 2020-02-06:
> To be fair, I am trying to build gnuradio, and clicked-plugins is listed as a dependency required to build gnuradio.

It's listed as a dependency because it is needed by `gr_modtool`

https://github.com/gnuradio/gnuradio/blob/v3.8.0.0/gr-utils/CMakeLists.txt#L33-L37
https://github.com/gnuradio/gnuradio/blob/v3.8.0.0/gr-utils/python/modtool/README.modtool#L7

In case you don't need `gr_modtool` you can try removing `- python-click-plugins` from `~/.pybombs/recipes/gr-recipes/gnuradio.lwr`