gnuradio/gnuradio #3188: Sanity for GnuradioConfigVersion.cmake

marcusmueller, 2020-02-09:
That is the file CMake executes when someone does a `find_package(Gnuradio "VERSION.VERSION")`. 

#3187 highlighted that we're not doing a good job there: Of course, an OOT module can't know whether 3.9-prerelease versions are compatible with what 3.9 will be, nor can it know whether they're compatible with 3.8.

So, a sane environment would allow the OOT author to allow "hey, I want at least GNU Radio 3.8.1.2", because they need a specific API (3.8) and they need a specific bug fix to be already present. But if there's 3.9.0.0 on the system, CMake shouldn't fail at that point. 

The solution I'd propose would be to change GnuradioConfigVersion.cmake.in to:

* set `PACKAGE_VERSION_COMPATIBLE` on
```REQUESTED MAJOR==GR MAJOR && GR VERSION >= REQUESTED VERSION```
* set `PACKAGE_VERSION_EXACT` on 
```GR VERSION == REQUESTED VERSION```
(all digits specified, not just the first three)

smunaut, 2020-02-09:
Note that the changes I was proposing (for which argilo made the PR), were _not_ meant to make it in release 3.9.

My PoV is that until 3.9 is released and is "guaranteed API" released, version checks make little sense at all since it might break at any time depending on what gets merged to master.

And so the cmake file in the "git version" of GR should be tolerant and declare itself 3.8 compatible during the dev phase, with the assumption that user of an unstable branch know what they're doing.

When 3.9 is _actually_ released and tagged (or the release candidate is), then version check should go to "strict" again and at that point the OOT peole need to go, check their OOT against 3.9 API and accept it as a supported version.

smunaut, 2020-02-09:
BTW that's actually similar to what most project do. When working on the "next version" they usually declare themselves as like 3.8.9999  and not 3.9 for that exact reason ... then only switch the actual git version to 3.9 at release time.

marcusmueller, 2020-02-09:
* You're right, no sense in using version checks against something that's not a version
* So, if you're consumer of an unreleased version of anything, you'll have to live with not being able to rely on a version check
* Therefore, it's not upstream's responsibility to try and guess what you think should be covered under compatibility. Current `master` is not compatible with 3.8, full stop. Otherwise, we wouldn't be developing it on master, but `maint-3.8`, which we dedicatedly decided *against*, for very good reason.

So, how to implement your recommendation:

> When 3.9 is actually released and tagged (or the release candidate is), then version check should go to "strict" again and at that point the OOT peole need to go, check their OOT against 3.9 API and accept it as a supported version.

... without special-casing for every release?

Problem: CMake's `find_package` has no useful way of doing that; you can only specify a version, not a range. So, if you want to check for any GR in say `{3.8.1.4...3.10.UINT_MAX.UINT_MAX}`, you'll need to go in and 

```
find_package(Gnuradio "3.10")
if(not found)
    find_package(Gnuradio "3.9")
    if(not found)
        find_package(Gnuradio "3.8")
        if(found)
            check(VERSION_ABI > 1 || (VERSION_ABI == 1 && VERSION_PATCH >= 4))
        endif()
    endif()
endif()
```

I doubt many of our OOT module maintainers are diligent enough to do that for more than 2 future minor releases.

That's where I'm coming from: maybe we should only allow the maintainers that actually have the time to care make sure that the GR version 100% works before they attempt a build.

For the others, it'd be fine to have a `check_version_in_range(version, min[, max])` macro, for usage like:

```
find_package(Gnuradio "3.8.1.4" REQUIRED) // still succeeds if version > 3.8.14
check_version_in_range(${GNURADIO_VERSION}, "3.8.1.4", "3.10")
if(${VERSION_CHECK_OK})
…
```

where having `-git` in the passed version indicates "this version is OK, whatever the required versions are, print warning that this is a development version", and "-rcN" are treated like the final release.

Would that sound like a good idea, @smunaut?

marcusmueller, 2020-02-09:
BTW, BTW:

> BTW that's actually similar to what most project do. When working on the "next version" they usually declare themselves as like 3.8.9999 and not 3.9 

I think it's a split, to be honest: the KiCAD does that, boost doesn't, the Linux kernel handles that differently alltogether, glibc agrees with you, zsh with me...

argilo, 2020-02-09:
> So, how to implement your recommendation [...] without special-casing for every release?

Perhaps @smunaut's suggestion could be implemented by returning `PACKAGE_VERSION_COMPATIBLE` if the requested version matches the one indicated by `git describe`. For master it's currently `v3.8.0.0-283-g38cdeafdff` but would go up to `v3.9.something` once a release candidate is tagged.

argilo, 2020-02-09:
As an OOT maintainer, I also think @marcusmueller's initial suggestion would also be fine. If I ask for GNU Radio 3.8 and you have master or 3.9, then it's better to attempt a build than to bail out immediately. If 3.9 turns out to be incompatible with my OOT I can always add `EXACT` to my `find_package`.

argilo, 2020-02-19:
@marcusmueller The CMake wiki helpfully provides an [example](https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/How-to-create-a-ProjectConfig.cmake-file#the-foobarfoobarconfigversioncmakein-file) that does exactly what you suggested. I put that into #3218.

closed 2020-04-10 by marcusmueller