gnuradio/gnuradio #3192: gnuradio-runtime / tpb_thread_body causing hardware overflows for source blocks

mchlwntz, 2020-02-11:
If a source block that uses a hardware driver isn't producing any items yet, for example a USRP source set to receive at some time in the future, its work function will only be called every 250 ms. This seems like a very long time, and it means that once samples start streaming from the hardware the runtime will very likely be stuck waiting instead of emptying buffers. 

I first noticed the issue receiving 2 channels from a B210, no matter the sample rate I would always get overflows at startup. I've since encountered it in a number of applications that never had this behavior before updating to 3.8.

I traced it back to commit 254fe5e which changed the logic in the BLKD_IN case of tpb_thread_body.cc. I've reverted this file and verified it fixes the problem.

More detail posted to the mailing list a few weeks ago: https://lists.gnu.org/archive/html/discuss-gnuradio/2020-01/msg00100.html

michaelld, 2020-02-11:
Do you mean specifically the changes to `tpb_thread_body.cc` starting on line 150: https://github.com/gnuradio/gnuradio/commit/254fe5e89403d4de1fa6663d09efdf946996aff3#diff-2a6e3c6c46e07604b86d096baf29694cL150 ... I mean, that's an enormous changeset!

If you change this value to 25 ms instead of 250 ms, you should see the same basic result as reverting the commit to that file. The basic issue as you note is that 250 ms is just too long to wait for some blocks to check for data again. Ideally this parameter would be controllable by the final block constructor somehow.

mchlwntz, 2020-02-12:
Yes, I think more specifically from this: https://lists.gnu.org/archive/html/commit-gnuradio/2016-07/msg00019.html

Out of curiosity, where does the 25 ms come in? I was under the impression it would get polled much faster than that. And yes, it would be nice to have control over that parameter. 

As a temporary workaround in some of my applications, I'm setting a recv timeout for UHD so work will only be called once and recv will block until there is data.

marcusmueller, 2024-01-08:
Going to close this, as we've been through really a lot of a ride regarding this issue.

One trick is to "wake" the block executor with a PMT message posted to its system port.

closed 2024-01-08 by marcusmueller