gnuradio/gnuradio #3194: Custom OOT block in 3.8 python error

SpontaneousDuck, 2020-02-11:
I am trying to build my own simple custom block in GR 3.8 with gr_modtool and have hit a roadblock trying to use it. Both when I try and use the block with ```make test``` and in python I get the error ```module 'test38_swig' has no attribute 'msgPulse_mm'```.

I am following the tutorial [here](https://wiki.gnuradio.org/index.php/OutOfTreeModules#Build_Tree_vs._Install_Tree) and trying to make my own version of the GNU Radio message strobe block. I created all the templates with ```gr_modtool``` and added the code. For right now I have exactly copied that block's code, just with a different block name. It appears to compile with no errors. I have also tried multiple other blocks and have the same problem. I installed GR 3.8 with pybombs and am able to use it otherwise just fine. I have linked the code I am using below. Thanks!

[code](https://github.com/SpontaneousDuck/gr-test38)

velichkov, 2020-02-19:
Hi @SpontaneousDuck,

Your OOT module is called `msgStrobe_mm` and there is no `msgPulse_mm`  block in your repo.  You need to use the correct name in your test, try the following
```diff
diff --git a/python/qa_msgStrobe_mm.py b/python/qa_msgStrobe_mm.py
index be75116..2a5b2ee 100755
--- a/python/qa_msgStrobe_mm.py
+++ b/python/qa_msgStrobe_mm.py
@@ -21,6 +21,7 @@
 
 from gnuradio import gr, gr_unittest
 from gnuradio import blocks
+import time
 import pmt
 import test38_swig as test38
 
@@ -35,7 +36,7 @@ class qa_msgStrobe_mm(gr_unittest.TestCase):
     def test_001_t(self):
         text = "TESTING"
         msg = pmt.intern("TESTING")
-        src = test38.msgPulse_mm(msg)
+        src = test38.msgStrobe_mm(msg)
         snk = blocks.message_debug()
 
         self.tb.msg_connect(src, "strobe", snk, "store")
@@ -49,6 +50,4 @@ class qa_msgStrobe_mm(gr_unittest.TestCase):
 
 
 if __name__ == '__main__':
-    print 'Blocked waiting for GDB attach (pid = %d)' % (os.getpid(),)
-    raw_input('Press Enter to continue: ')
     gr_unittest.run(qa_msgStrobe_mm)
```

You may post questions or problems like this to the [mailing list](https://wiki.gnuradio.org/index.php/MailingLists) or [IRC/Slack channels](https://wiki.gnuradio.org/index.php/IRC)

SpontaneousDuck, 2020-02-24:
Perfect! Now I feel super dumb haha. Thanks for the help! Will definitely check my naming carefully iin the future.

SpontaneousDuck, 2020-02-26:
So everything works great when I have only one block but as soon as I add a second block to the repository with ```gr_modtool```, I am unable to import the swig library. Using ```nm``` I can see that the functions from the .cpp file for the second block is not being added to the ```_test38_swig.so ``` output file. this is causing swig to be unable to find symbols when it is run. If I comment out the lines in the swig ```*.i``` config file that add the second header file, everything works again. I have committed the small changes I made to the repo above. Thanks for all your help!

velichkov, 2020-02-26:
> So everything works great when I have only one block but as soon as I add a second block to the repository with `gr_modtool`, I am unable to import the swig library. 

I can't reproduce this. Just did a clean rebuild from `f826f78b1d919ce22df3829a874a53d24b1c3755`

> Using `nm` I can see that the functions from the .cpp file for the second block is not being added to the `_test38_swig.so ` output file.

They seems to be added, here are the first several lines.
```
$ nm --demangle ./swig/_test38_swig.so  | grep sync_test 
0000000000014420 t _p_gr__test38__sync_testTo_p_gr__basic_block
00000000000143b0 t _p_gr__test38__sync_testTo_p_gr__block
0000000000014450 t _p_gr__test38__sync_testTo_p_gr__sync_block
0000000000014d70 t sync_test_sptr_swigregister
0000000000014de0 t sync_test_swigregister
00000000000186e0 t _wrap_delete_sync_test
000000000001c4f0 t _wrap_delete_sync_test_sptr
000000000001f420 t _wrap_new_sync_test_sptr
000000000001276c t _wrap_new_sync_test_sptr.cold.478
000000000001f320 t _wrap_sync_test_make
.....
```
```
$ nm --demangle ./lib/libgnuradio-test38.so  | grep sync_test 
000000000000ce40 t _GLOBAL__sub_I_sync_test_impl.cc
00000000000191e0 t gr::test38::sync_test_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&)
0000000000019360 t gr::test38::sync_test_impl::sync_test_impl()
000000000000cd54 t gr::test38::sync_test_impl::sync_test_impl() [clone .cold.279]
0000000000019300 t gr::test38::sync_test_impl::sync_test_impl()
0000000000019270 t gr::test38::sync_test_impl::~sync_test_impl()
0000000000019230 t gr::test38::sync_test_impl::~sync_test_impl()
0000000000019330 t gr::test38::sync_test_impl::~sync_test_impl()
...
```
> this is causing swig to be unable to find symbols when it is run. 

the `ldd -r ./swig/_test38_swig.so` also does not report any undefined symbols.

> If I comment out the lines in the swig `*.i` config file that add the second header file, everything works again. I have committed the small changes I made to the repo above. Thanks for all your help!

My recommendation is to do a clean rebuild and in case you've installed your OOT module then first uninstall it and then do a clean rebuild. By clean rebuild I mean removing the whole build directory and starting from scratch.

SpontaneousDuck, 2020-02-26:
So it is getting added! The problem I am seeing is with the ```make()``` function (```_ZN2gr6test389sync_test4makeEv```). that one is showing a U with ```nm```. ```ctest -V``` is giving me the error that wig has undevined symbol ```_ZN2gr6test389sync_test4makeEv```. Thanks!!

velichkov, 2020-02-26:
``` 
$ nm ./swig/_test38_swig.so  | grep _ZN2gr6test389sync_test4makeEv
                 U _ZN2gr6test389sync_test4makeEv
$ nm ./lib/libgnuradio-test38.so | grep _ZN2gr6test389sync_test4makeEv
00000000000194d0 T _ZN2gr6test389sync_test4makeEv
000000000000cd90 t _ZN2gr6test389sync_test4makeEv.cold.280
```
Yes, it is `undefined` in the `_test38_swig.so` as it should because it is implemented in `libgnuradio-test38.so` where it is not undefined.

As I said try uninstalling (`sudo make uninstall`) and then a clean rebuild. If it it is still undefined provide the full output of
```
pwd
ldd -r ./lib/libgnuradio-test38.so
ldd -r ./swig/_test38_swig.so
```

SpontaneousDuck, 2020-02-27:
Ahhh, so I never ran ```sudo make install```as the documentation I mentioned above about OOT said you did not need to do that until done with testing ```make test```. After running install, the test worked properly! Is that a bug in the Cmake files or just incorrect doumentation? Thanks for your help!

velichkov, 2020-02-27:
> Ahhh, so I never ran `sudo make install`as the documentation I mentioned above about OOT said you did not need to do that until done with testing `make test`. After running install, the test worked properly! Is that a bug in the Cmake files or just incorrect documentation? Thanks for your help!

The documentation is correct that you should not install until everything is ready and also it's really convenient to be able to develop and test a new version (that contains new blocks) without installing it (or uninstalling the previous version).

Which Linux distribution do you use and which version? Which cmake version do you use? 

Go in the `build` directory and give me the full output of the following commands.
```
$ grep RPATH CMakeCache.txt
$ objdump -p ./swig/_test38_swig.so | grep RPATH
$ ldd -r ./swig/_test38_swig.so  | grep test38
```
And read also [RPATH handling](https://gitlab.kitware.com/cmake/community/-/wikis/doc/cmake/RPATH-handling)

SpontaneousDuck, 2020-02-27:
I am using Linux Mint 19.3 which is based on Ubuntu 18.04 I believe. Cmake is version ```3.10.2``` which is the newest on Ubuntu's package repo for 18.04. 

```
$ grep RPATH CMakeCache.txt
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
//ADVANCED property for variable: CMAKE_SKIP_INSTALL_RPATH
CMAKE_SKIP_INSTALL_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_RPATH
CMAKE_SKIP_RPATH-ADVANCED:INTERNAL=1
```
```$ objdump -p ./swig/_test38_swig.so | grep RPATH``` Returns nothing
```
$ ldd -r ./swig/_test38_swig.so  | grep test38
libgnuradio-test38.so.1.0.0git => /home/ken/gnuradio/lib/libgnuradio-test38.so.1.0.0git (0x00007fe4565c8000)
```
Thanks!

velichkov, 2020-02-27:
> I am using Linux Mint 19.3 which is based on Ubuntu 18.04 I believe. Cmake is version `3.10.2` which is the newest on Ubuntu's package repo for 18.04.

It's a build old but don't update it for now as there is a chance that it's a problem in the cmake and first we need to find the root cause.

> ```
> $ grep RPATH CMakeCache.txt
> CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
> CMAKE_SKIP_RPATH:BOOL=NO
> //ADVANCED property for variable: CMAKE_SKIP_INSTALL_RPATH
> CMAKE_SKIP_INSTALL_RPATH-ADVANCED:INTERNAL=1
> //ADVANCED property for variable: CMAKE_SKIP_RPATH
> CMAKE_SKIP_RPATH-ADVANCED:INTERNAL=1
> ```

The output is exactly the same as on my Fedora 29 with cmake 3.14.5

> `$ objdump -p ./swig/_test38_swig.so | grep RPATH` Returns nothing

And here is the problem, no rpath is being set in the _test38_swig.so
> ```
> $ ldd -r ./swig/_test38_swig.so  | grep test38
> libgnuradio-test38.so.1.0.0git => /home/ken/gnuradio/lib/libgnuradio-test38.so.1.0.0git (0x00007fe4565c8000)
> ```
so when the test loads it loads the previous version of the library from your prefix and not the newest version from the `build` directory. For reference here is the output from my system.
```
$ objdump -p ./swig/_test38_swig.so  | grep RPATH
  RPATH                /home/vasko/gr38/src/gr-test38/build/lib:
$ ldd -r ./swig/_test38_swig.so | grep test38
	libgnuradio-test38.so.1.0.0git => /home/vasko/gr38/src/gr-test38/build/lib/libgnuradio-test38.so.1.0.0git (0x00007f8618e4a000)
```
The .so has an RPATH header and as a result loads the library from the `build` directory.

Now I'm going to setup a clean `Linux Mint 19.3` container and test in it.

velichkov, 2020-02-27:
In my container the `_test38_swig.so` does not have `RPATH` but has `RUNPATH`
```
  RUNPATH              /root/gr-test38/build/lib:
```
That's something new for me. Give me the full output of the following commands.
```
objdump -p ./swig/_test38_swig.so
cat ./swig/CMakeFiles/test38_swig.dir/link.txt
grep -R rpath . 
grep -R runpath .
g++ --version
c++ --version
```
From https://wiki.debian.org/RpathIssue

> Dynamic linking
>
>The job of the dynamic linker and loader (ld.so) is to resolve the executable's dependencies on shared libraries and to load the required ones at run-time. It relies on the executable's NEEDED headers to find the shared libraries with a matching SONAME (which includes the library name and ABI version). 
>
> The dynamic linker will look for a matching library in the following locations, in this order, which can be changed (see the footnotes below):
>
> 1. the DT_RPATH dynamic section attribute of the library causing the lookup
> 2. the DT_RPATH dynamic section attribute of the executable
> 3. the LD_LIBRARY_PATH environment variable, unless the executable is setuid/setgid.
> 4. the DT_RUNPATH dynamic section attribute of the executable
> ...

You **do** have `LD_LIBRARY_PATH` set as you are using an installation in a custom prefix (you are sourcing the `setup_env.sh` file) and it has precendence over `DT_RUNPATH` and that's why the older installed version gets used instead of the local pointed by the `RUNPATH`.

Now we need to find a way to force an `RPATH` header to be added and not a `RUNPATH` for all linux distributions.

Read also http://blog.tremily.us/posts/rpath/

SpontaneousDuck, 2020-02-27:
I am seeing the same thing with ```RUNPATH```
```
$ objdump -p ./swig/_test38_swig.so

./swig/_test38_swig.so:     file format elf64-x86-64

Program Header:
    LOAD off    0x0000000000000000 vaddr 0x0000000000000000 paddr 0x0000000000000000 align 2**21
         filesz 0x0000000000042a90 memsz 0x0000000000042a90 flags r-x
    LOAD off    0x0000000000042eb0 vaddr 0x0000000000242eb0 paddr 0x0000000000242eb0 align 2**21
         filesz 0x0000000000004880 memsz 0x0000000000004e80 flags rw-
 DYNAMIC off    0x0000000000043bf8 vaddr 0x0000000000243bf8 paddr 0x0000000000243bf8 align 2**3
         filesz 0x0000000000000250 memsz 0x0000000000000250 flags rw-
    NOTE off    0x00000000000001c8 vaddr 0x00000000000001c8 paddr 0x00000000000001c8 align 2**2
         filesz 0x0000000000000024 memsz 0x0000000000000024 flags r--
EH_FRAME off    0x000000000003af60 vaddr 0x000000000003af60 paddr 0x000000000003af60 align 2**2
         filesz 0x0000000000000bc4 memsz 0x0000000000000bc4 flags r--
   STACK off    0x0000000000000000 vaddr 0x0000000000000000 paddr 0x0000000000000000 align 2**4
         filesz 0x0000000000000000 memsz 0x0000000000000000 flags rw-
   RELRO off    0x0000000000042eb0 vaddr 0x0000000000242eb0 paddr 0x0000000000242eb0 align 2**0
         filesz 0x0000000000001150 memsz 0x0000000000001150 flags r--

Dynamic Section:
  NEEDED               libgnuradio-test38.so.1.0.0git
  NEEDED               libpython3.6m.so.1.0
  NEEDED               libgnuradio-runtime.so.3.8.0
  NEEDED               libboost_system.so.1.65.1
  NEEDED               libpthread.so.0
  NEEDED               librt.so.1
  NEEDED               libstdc++.so.6
  NEEDED               libgcc_s.so.1
  NEEDED               libc.so.6
  RUNPATH              /home/ken/gr-test38/build/lib:
  INIT                 0x000000000000e778
  FINI                 0x000000000002e1a0
  INIT_ARRAY           0x0000000000242eb0
  INIT_ARRAYSZ         0x0000000000000010
  FINI_ARRAY           0x0000000000242ec0
  FINI_ARRAYSZ         0x0000000000000008
  GNU_HASH             0x00000000000001f0
  STRTAB               0x00000000000028f8
  SYMTAB               0x00000000000006c0
  STRSZ                0x00000000000042c4
  SYMENT               0x0000000000000018
  PLTGOT               0x0000000000244000
  PLTRELSZ             0x0000000000001248
  PLTREL               0x0000000000000007
  JMPREL               0x000000000000d530
  RELA                 0x0000000000006fa8
  RELASZ               0x0000000000006588
  RELAENT              0x0000000000000018
  VERNEED              0x0000000000006e98
  VERNEEDNUM           0x0000000000000005
  VERSYM               0x0000000000006bbc
  RELACOUNT            0x0000000000000345

Version References:
  required from libgcc_s.so.1:
    0x0b792650 0x00 13 GCC_3.0
  required from librt.so.1:
    0x09691a75 0x00 11 GLIBC_2.2.5
  required from libpthread.so.0:
    0x09691a75 0x00 04 GLIBC_2.2.5
  required from libstdc++.so.6:
    0x0297f865 0x00 12 GLIBCXX_3.4.15
    0x02297f89 0x00 10 GLIBCXX_3.4.9
    0x056bafd3 0x00 07 CXXABI_1.3
    0x0297f871 0x00 06 GLIBCXX_3.4.21
    0x08922974 0x00 03 GLIBCXX_3.4
  required from libc.so.6:
    0x0d696914 0x00 09 GLIBC_2.4
    0x06969194 0x00 08 GLIBC_2.14
    0x09691a75 0x00 05 GLIBC_2.2.5
    0x09691974 0x00 02 GLIBC_2.3.4

$ cat ./swig/CMakeFiles/test38_swig.dir/link.txt
/usr/bin/c++ -fPIC -O3 -DNDEBUG  -shared  -o _test38_swig.so CMakeFiles/test38_swig.dir/CMakeFiles/test38_swig.dir/test38_swigPYTHON_wrap.cxx.o -Wl,-rpath,/home/ken/gr-test38/build/lib: ../lib/libgnuradio-test38.so.f826f78b /usr/lib/x86_64-linux-gnu/libpython3.6m.so /home/ken/gnuradio/lib/libgnuradio-runtime.so.3.8.0.0 /home/ken/gnuradio/lib/libgnuradio-pmt.so.3.8.0.0 /home/ken/gnuradio/lib/libvolk.so.2.0 -ldl -lm /usr/lib/x86_64-linux-gnu/libboost_program_options.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/lib/x86_64-linux-gnu/liblog4cpp.so /usr/lib/x86_64-linux-gnu/libgmpxx.so /usr/lib/x86_64-linux-gnu/libgmp.so -lrt

$ g++ --version
g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

$ c++ --version
c++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
```
```grep -R rpath . ``` returned same as the cat call. ```grep -R runpath .``` returned nothing.

velichkov, 2020-02-27:
I found a workaround, you need to link with `-Wl,--disable-new-dtags` flag.
```
rm CMakeCache.txt
LDFLAGS="-Wl,--disable-new-dtags" cmake ..
make
```
It looks like in Debian and Debian based distributions this new dtags feature is enabled by default and the linker sets RUNPATH header instead of RPATH but as described above this is problematic when using a gnuradio installed in a custom prefix.

You may also consider adding `export LDFLAGS="-Wl,--disable-new-dtags"` in your setup_env.sh

SpontaneousDuck, 2020-02-28:
Appears to work! I added it to my setup_env.sh  and am able to test from clean build without installing first. Should we document this somewhere for other people who may come across the problem?

velichkov, 2020-02-28:
> Appears to work! 

:rocket: 

> I added it to my setup_env.sh and am able to test from clean build without installing first. Should we document this somewhere for other people who may come across the problem?

Yes, you can register an account in the wiki](https://wiki.gnuradio.org/) and add it in the tutorial or propose this in the [slack #docs channel](https://slack.gnuradio.org/).

SpontaneousDuck, 2020-02-28:
Thanks for all your help @velichkov ! Hope this helps anyone else coming along with the problem. I'll try and get this added to docs as well.

closed 2020-02-28 by SpontaneousDuck