gnuradio/gnuradio #3196: GRC: Error in FFT block for GNURadio 3.8

lunar-mosiur, 2020-02-11:
Problems in issue [#2158](https://github.com/gnuradio/gnuradio/issues/2158) comes back again in GR 3.8. 

Fixes in [#1925](https://github.com/gnuradio/gnuradio/pull/1925) is already implemented in GR 3.8. However still the problem exists like before.
[image]

drmpeg, 2020-02-11:
That's an OOT block. You have to submit an issue to https://github.com/ghostop14/gr-clenabled

devnulling, 2020-02-11:
Tagging @ghostop14 , but I'm going to close this issue.

closed 2020-02-11 by devnulling

ghostop14, 2020-02-12:
Fixed in clenabled.  Just pushed an updated yml.  This crops up every once in a while when gr_modtool converted xml to yml going from 3.7 to 3.8, it ignored the real variable name and always wrote it as vlen.  Not sure if that's been fixed yet or not.