gnuradio/gnuradio #3197: Flow graph not execute in GRC v3.8 on Windows 10

delimaras, 2020-02-12:
I'm new in gnuradio and I can't to run my flow graph.

After generating and in executing process, my flow graph in GRC 3.8 (Windows version) not running and terminal return this:

**Done (return code -1073741795)**

Where is the problem?

Thank You.

marcusmueller, 2020-02-13:
Please add clarification on how you've installed. I'm afraid we can't debug this based on your description, so I'll have to ask you to attach a debugger to your python instance to get a start at this. Please do add the requested info and afterwards reopen the issue. Thank you!

closed 2020-02-13 by marcusmueller

tinyxuyan, 2020-08-11:
Does anyone solve the issue? I met the same issue in windows 10.
Installation methods: download the [ 'gnuradio_3.8.1.0_win64.msi'](http://www.gcndevelopment.com/gnuradio/downloads.htm) and directly install it.
When I executed the simple test flow graph as  @delimaras mentioned, I got the error [Known Windows Build Issues](https://wiki.gnuradio.org/index.php/WindowsInstall). After solving the error according to the solution, there was no error but the terminal returned `Done (return code -1073741795)` and displayed nothing.

delimaras, 2020-08-12:
I have found a temporary solution with installed the older version 3.8.0.0. May this solve your issue. Something going wrong with 3.8.1.0 version.

vaughanatworld, 2021-01-10:
I'm seeing the same problem with 3.8.1.0 Also I'm a newbie.
It has something to do with trying to execute an illegal instruction.

vaughanatworld, 2021-01-11:
I'm running an AMD A4-3300M with win10. I wonder if there are SIMD instruction incompatibilities between Intel and AMD with this CPU vintage.